norlab-ulaval/libpointmatcher #199: Set to Subset matching?

antithing, 2017-03-28:
Hi, and thank you for making this code available. 

I have two point clouds, where one is smaller, and contains a subset of the points from the other.

Will libpointmatcher work in this case? Or does it expect two clouds of the same size?

Thanks again.

pomerlef, 2017-03-28:
Unequal point cloud size is the usual case for most of the applications.

If your reading point cloud is **fully** contained in the reference, you can assume that all points should match to a reference point. This is the easy case as your overlap is around 100%.

If your reading point cloud is **partially** contained in the reference, you need to use the outlier filters to identify which paired points should not be considered for the minimization.

antithing, 2017-03-28:
Great! Thank you.

antithing, 2017-03-29:
@pomerlef, I have the lib up and running, and am using it on my datasets, but getting pretty bad results. If you have a moment, could you let me know what is happening here? The attached image is the result when i run the icp_customized example on the attached zipped data. The points are from a stereo camera mapping, a global map, and a per-frame map.  What can i look at to improve these results? or is it my data that is the issue?

Thank you!

[data.zip](https://github.com/ethz-asl/libpointmatcher/files/880254/data.zip)

[image]

pomerlef, 2017-05-03:
Stereo data are very noisy compared to laser point cloud. The data tends to generate a lot of wrong depth. Typically, people would run a registration algorithm based on visual features and then, if you're good in your noise identification, refine it with geometric registration. More often than not, the second step is dropped.

svdalv0313, 2017-05-08:
Hi,I  meet the same problems of Set-to-Subset point cloud matching now ,have you solved it?

antithing, 2017-05-08:
I have not, i am afraid. I got an ok result if the clouds where very close together, but it was never good enough to actually use. I think my clouds were just too different.

svdalv0313, 2017-05-08:
did you do  rough registration  at first?do you finish registration in 30ms? why do you not use PCL to do this ,could I communicate with you by mail? my email is suqiang0313@gmail.com,Thanks

svdalv0313, 2017-05-08:
Hi ,I have met the same problem now ,I want to communicate with you ,did
you do this with the method in PCL tutorials like  Robust pose estimation
of rigid objects

pomerlef, 2017-05-08:
PCL is a great library too. We don't cover the robust estimation in our tutorials (yet), but we deal with robust estimation by using outlier filters. The current list is (taken from `./examples/list_modules`):

- `GenericDescriptorOutlierFilter`: This filter weights matched points based on a 1D descriptor of either a single point cloud (either the reference or the reading). The descriptor values must be larger than zero.
- `MaxDistOutlierFilter`: This filter considers as outlier links whose norms are above a fix threshold.
- `MedianDistOutlierFilter`: This filter considers as outlier links whose norms are above the median link norms times a factor. Based on [3].
- `MinDistOutlierFilter`: This filter considers as outlier links whose norms are below a threshold.
- `NullOutlierFilter`: Does nothing.
- `RobustWelschOutlierFilter`: Robust weight function part of the M-Estimator familly. The Welsch weight uses an exponential decay reducing the influence of matched point farther away [4]. More explicitly, the function is w = exp[- (matched distance)^2/scale^2].
- `SurfaceNormalOutlierFilter`: Hard rejection threshold using the angle between the surface normal vector of the reading and the reference. If normal vectors or not in the descriptor for both of the point clouds, does nothing.
- `TrimmedDistOutlierFilter`: Hard rejection threshold using quantile. This filter considers as inlier a certain percentage of the links with the smallest norms. Based on [5].
- `VarTrimmedDistOutlierFilter`: Hard rejection threshold using quantile and variable ratio. Based on [6].

It will be up to you to read the articles to understand more what they do.

svdalv0313, 2017-05-08:
ok ,thank you very much

closed 2017-08-07 by pomerlef

bxc237, 2019-01-10:
Hi @suqiang0313, have you solved the Set-to-Subset issue? I encounter the same issue #309. Hi @pomerlef, it would be a great help if you could provide more details about how to tackle the robust estimation issue using libpointmatcher? THX...