norlab-ulaval/libpointmatcher #205: Icp to improve point cloud registration

v4ven27, 2017-04-28:
Hello,

I'm using libpointmatcher_ros to improve registration of successive point clouds obtained from a LIDAR scanner. Point clouds have already been transformed to a global reference frame. 

Here's an example input point cloud:
[image]

Here's an example of registration without ICP:
[image]

Here's an example of registration after ICP with libpointmatcher:
[image]

As seen above, I'm getting poor results. Please note that I'm using the default ICP mode. Am I missing something? Should I be using a custom ICP pipeline? 

Please let me know if you need more information.

Thanks!
Ven

pomerlef, 2017-05-01:
The basic pipeline is there to work with the example scans, but rarely work in general. The library is there for people to build their own registration solution, which is often custom to their applications.

v4ven27, 2017-05-02:
Hi Francois, 

Thanks for your answer. How would you recommend building a registration solution for a LIDAR scan of a simple roof? For eg,

[image]

The registration results, without ICP, already look pretty good. I have reasonably accurate pose estimates which I'd like to use to initialize ICP similar to [ethzasl_icp_mapping issue#16](https://github.com/ethz-asl/ethzasl_icp_mapping/issues/16) . I tried customizing the ICP pipeline, after going through the tutorials, file but I couldn't get it to work. 

What's the best way to go about designing a custom registration pipeline for such objects? How do I select a good set of filters for my application? 

Thanks!

pomerlef, 2017-05-03:
From the screenshot, your reconstruction seems already good. I would be surprise if a quick ICP setup would make it better. For a more _advance_ solution you would need:

- a good knowledge on your sensor noise
- a good knowledge on your prior noise
- a good knowledge on your velocity model
- a way to fuse those information with a _registration solution_

Libpointmatcher only provides a framework for different registration solutions, which can be used inside an application. The executable that we provide are meant to be demonstration of the capabilities of the library, not a final solution.

As for many issues opened here, we cannot support application specific questions. We too many requests and problems are very specific to the data provided.

closed 2017-08-07 by pomerlef