norlab-ulaval/libpointmatcher #213: Install rule issue

artivis, 2017-06-06:
Hi,

I have a Travis job for a package that depends on `libpointmatcher`  (`ros-indigo-libpointmatcher`) which fails building with the following error :  
```terminal
No rule to make target `/tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/contrib/yaml-cpp-pm/include'
```

The Travis job can be found [here](https://travis-ci.org/artivis/laser_odometry_libpointmatcher/jobs/236428446).

Any idea ?

pomerlef, 2017-06-07:
Was is working before and now it fails? Is so, it would help to know when it crashes to track the problem.

artivis, 2017-06-08:
No it never worked as it is a job that I very recently set up. I'll see if I can reproduce the error on my machine and/or get more detailed info.

artivis, 2017-06-22:
I couldn't reproduce exactly the same error logs on my machine but it might be due to the fact that I also depends on [`pointmatcher_ros`](https://github.com/ethz-asl/pointmatcher-ros) which misses a dependency on `libpointmatcher`.  
See [`pointmatcher_ros` PR 1](https://github.com/ethz-asl/pointmatcher-ros/pull/1).

closed 2018-09-10 by simonpierredeschenes