norlab-ulaval/libpointmatcher #214: Problem in point cloud positioning with the Introduction to ICP

dydis, 2017-06-07:
Hi,
I am following the introduction to [ICP tutorial](http://libpointmatcher.readthedocs.io/en/latest/ICPIntro/). After running _pmicp_ with the default config file (found in examples/icp_tutorial/icp_tutorial_cfg.yaml), I noticed (in Paraview) that my reference point cloud (cloud_0.vtk) is not on the same coordinates as the reference point cloud generated by the algorithm (vissteps-reference). **Attached a screenshot of the problem**.
Can you please help me understand the problem?
[image]

pomerlef, 2017-06-08:
Great question. Quick answer, it's normal.

The output given by the vtk inspector (visstep) is an internal view of what's going on inside the iterative process. We translate both point clouds to have the origin of the reference point cloud to its center of masse. We do that because we solve the rotation error around the origin and some point clouds (terrestrial survey, for example) use global coordinates with the origin being excessively far away. We do the transformation inverse before outputting the final transformation.

In your case, this means that cloud*.vtk are coherent together and visstep*.vtk are in a different frame but coherent together.

dydis, 2017-06-20:
I get it. Thank you for your help.

closed 2017-06-20 by dydis