norlab-ulaval/libpointmatcher #215: Reflection Matrix in PointToPointCovErrorMinimizer

pierrefranklin, 2017-08-07:
Taken directly from the code in PointToPointErrorMinimizer::compute(...):

```
// It is possible to get a reflection instead of a rotation. In this case, we
// take the second best solution, guaranteed to be a rotation. For more details,
// read the tech report: "Least-Squares Rigid Motion Using SVD", Olga Sorkine
// http://igl.ethz.ch/projects/ARAP/svd_rot.pdf
```

The code around this (lines 220 to 234) is missing from the PointToPointCovErrorMinimizer (lines 333 to 340) in ErrorMinimizersImpl.cpp.

pomerlef, 2017-08-07:
Good catch. That gives me another argument to start refactoring `PointToPoint` functions as it happen with `PointToPlane`.

I'll try to do that sooner that later.

artivis, 2017-08-12:
Any chance to see those lines simply copied while waiting for the refactoring :) ??

pomerlef, 2017-08-12:
Yes, it would make sense. I'm not installed to check that on my side.

You can send a pull request if you feel comfortable with it. Otherwise, @davidlandry93 could you have a look?

artivis, 2017-08-14:
see PR #217

pomerlef, 2017-08-24:
Merged, thanks @artivis!

closed 2017-08-24 by pomerlef