norlab-ulaval/libpointmatcher #219: Question regarding creating your own transformation

Olimoyo, 2017-08-29:
Hi,
I've been using Pointmatcher for rigid transformations and it's working really well for my current application. 
To extend my project, I want to use a custom transformation which is not (4x4) but (4x7) applied to point features (x,y,z,1) and three descriptors (d1,d2,d3). Would this be possible with the current state of the library?

Thank you

pomerlef, 2017-08-29:
It should be possible without too much trouble. You need to define a new class inheriting from `Transformation`.

From the compiled API doc:

Public Member Functions of `Transformation`:

 Output | Function 
---|---
|  | Transformation () <br> Construct without parameter.
|  | Transformation (const std::string &className, const ParametersDoc paramsDoc, const Parameters &params) <br> Construct with parameters.|
virtual | ~Transformation () <br> virtual destructor
virtual DataPoints | compute (const DataPoints &input, const TransformationParameters &parameters) const =0 <br> Transform input using the transformation matrix.
virtual bool | checkParameters (const TransformationParameters &parameters) const =0 <br> Return whether the given parameters respect the expected constraints.
virtual TransformationParameters | correctParameters (const TransformationParameters &parameters) const =0 <br> Return a valid version of the given transformation.

There is also some information here:
http://libpointmatcher.readthedocs.io/en/latest/TransformationDev/

Olimoyo, 2017-08-31:
Thank you for the links!

Would I also need to implement my own error minimizer to solve for my new transformation?

pomerlef, 2017-09-01:
Yes, the minimization math are very specific to the transformation parameters.

Olimoyo, 2017-09-01:
Alright, that's what I thought. Thank you for the information!

closed 2017-09-01 by Olimoyo