norlab-ulaval/libpointmatcher #222: Is there a Python binding to libpointmatcher

s0r2637, 2017-10-03:
I was wondering if there is a Python binding to libpointmatcher?

pomerlef, 2017-10-03:
Clearly not, but I would highly appreciate it ;)

pomerlef, 2020-05-27:
We starting to work on Python binding. If someone has some inputs, advices or needs. Now is the time to tell us!

PhiBabin, 2020-05-30:
There was the start of an implementation of python bindings. It supported registration, add numpy array as pointcloud input and support extracting the debugging information that are normally in the .vtk output. However, it needs to be clean up, there is a bunch of code that was made specifically for debugging penalized ICP. https://github.com/norlab-ulaval/test_penalized_icp

YoshuaNava, 2020-06-02:
@PhiBabin How did you attempt the Python binding in your branch?

AlexandreG87, 2020-06-02:
@YoshuaNava Thanks for the informations! I'll take a look at them! 🙂

@PhiBabin I don't have access to the repo you've mentionned. 🙁

PhiBabin, 2020-06-03:
@YoshuaNava  I used boost.python.
@AlexandreG87 Oups, my bad, it is now public.

AlexandreG87, 2020-06-26:
Alright! So, after researching about all the Python bindings tools available, here are the ones I consider to have good potential to do the job. These solutions have been selected base on the following criteria :

### Primary criteria:

- Performance
- Modern C++ and Python3 compatibility
- Effective Numpy to Eigen conversion support (not 100% sure for cppyy)
- Template compatibility
- The ease of use and maintenance
- The solution must still be actively maintained
- No or minimal modification to the original code must be done

### Secondary criteria:

- Availability/Quality of documentation and tutorials
- Control and flexibility over interface implementation
- Does not require too much dependency

#### The fully automatic solutions :

- [**Automatic Wrapper and Interface Generator (AutoWIG)**](https://autowig.readthedocs.io/en/latest/) :
    - [Article on AutoWIG](https://peerj.com/articles/cs-149/)
    - [YouTube video explaning how AutoWIG works](https://youtu.be/N4q_Vud77Hw)
    - [GitHub repo](https://github.com/StatisKit/AutoWIG)

- [**cppyy**](https://cppyy.readthedocs.io/en/latest/) :
    - [Paper on cppyy](http://wlav.web.cern.ch/wlav/Cppyy_LavrijsenDutta_PyHPC16.pdf)
    - [Bitbucket repo](https://bitbucket.org/wlav/cppyy/src/master/)
    - [Tutorial](https://bitbucket.org/wlav/cppyy/src/master/doc/tutorial/CppyyTutorial.ipynb?viewer=nbviewer&fileviewer=notebook-viewer%3Anbviewer)
    - [Stack Overflow answer by the author](https://stackoverflow.com/questions/57862035/how-to-implement-python-interfaces-for-c-libraries)

#### And the semi-automatic solutions :

- [**Pybind11**](https://pybind11.readthedocs.io/en/master/) :
    - [GitHub repo](https://github.com/pybind/pybind11)
    - [JaggedVerge tutorial](http://www.jaggedverge.com/2016/09/generating-python-bindings-from-c11-with-pybind11/)
    - [Real Python tutorial](https://realpython.com/python-bindings-overview/#pybind11)
    - [YouTube playlist tutorial](https://www.youtube.com/playlist?list=PLvKAPIGzFEr8n7WRx8RptZmC1rXeTzYtA)
    - [Integrate Python and C++ with pybind11 - Robert Smallshire](https://youtu.be/YReJ3pSnNDo)

- [**Boost.Python**](https://www.boost.org/doc/libs/1_73_0/libs/python/doc/html/index.html) :
    - [GitHub repo](https://github.com/boostorg/python)
    - [Official tutorial](https://www.boost.org/doc/libs/1_73_0/libs/python/doc/html/tutorial/index.html)
    - [Developpez.com tutorial](https://cpp.developpez.com/tutoriels/interfacer-cpp-python/) (French tutorial)
    - [Integrating Python and C++ with Boost Python part 1 | SciPy 2014 | Austin Bingham](https://youtu.be/GE8EsGUsC2w)
    - [Integrating Python and C++ with Boost Python part 2 | SciPy 2014 | Austin Bingham](https://youtu.be/WLKuHbkh6jU)
    - [Integrating Python and C++ with Boost Python part 3 | SciPy 2014 | Austin Bingham](https://youtu.be/kR5tREIotlI)
    - [Integrating Python and C++ with Boost Python part 4 | SciPy 2014 | Austin Bingham](https://youtu.be/GgtlExpPM1c)

- [**Cython**](https://cython.org/)
    - [Cython docs](https://docs.cython.org/en/latest/index.html) :
    - [Cython GitHub repo](https://github.com/cython/cython)
    - [Cython Wiki](https://github.com/cython/cython/wiki)
    - [Real Python tutorial](https://realpython.com/python-bindings-overview/#cython)
    - [Paper](https://ieeexplore.ieee.org/document/5582062) (IEEE Xplore access required)

#### List of projects and examples using these tools :

- **AutoWIG** :
    - [StatisKit](https://github.com/StatisKit/StatisKit)
    - [FP17](https://github.com/StatisKit/FP17)

- **cppyy** :
    - [Example repos from the docs](https://cppyy.readthedocs.io/en/latest/examples.html)

- **Pybind11** :
    - [Tensorflow](https://github.com/tensorflow/tensorflow) (Use Eigen too)
    - [Pytorch](https://github.com/pytorch/pytorch) (Use Eigen too)
    - [NumpyEigen](https://github.com/fwilliams/numpyeigen)
    - [Open3D](https://github.com/intel-isl/Open3D) (Use Eigen too)
[... 5 lines cut ...]
- **Boost.Python** :
    - [EigenPy](https://github.com/stack-of-tasks/eigenpy)
    - [MiniEigen](https://github.com/eudoxos/minieigen/blob/master/src/minieigen.cpp)
    - [Examples](https://github.com/TNG/boost-python-examples)

- **Cython** :
    - [Eigency](https://github.com/wouterboomsma/eigency)
    - [SageMath](https://github.com/sagemath/sage)
    - [Numpy](https://github.com/numpy/numpy)
    - [python-pcl](https://github.com/strawlab/python-pcl)

#### Some more stuff on the subject :

- [Real Python bindings tutorial](https://realpython.com/python-bindings-overview/)
- [Blog - Which tool should you use? by Stephan Behnel (Cython core developper)](http://blog.behnel.de/posts/cython-pybind11-cffi-which-tool-to-choose.html) (Not just about Cython!)
- [Examples for different Python bindings tools](https://github.com/qingfengxia/python_wrap)
- [Calling C/C++ from Python? - Stack Overflow](https://stackoverflow.com/questions/145270/calling-c-c-from-python)

### In short...

Well, all of the above tools have their strenghs and weaknesses, none of them are perfect. But, I think it's worth trying the fully automatic solutions first, since they are designed with the ease of use and maintenance in mind. On the other hand, **AutoWIG** generate **Pybind11** or **Boost.Python** bindings, both of whom have extensive of tutorials, documentations and examples available. So, in the event **AutoWIG** doesn't  All of these tools are focused on performance, so I'm not worried about that too much. The main struggle being the Python call overhead, except for the **PyPy** interpreter. **cppyy**, **Pybind11** and **Cython** are compatible with the **PyPy** interpreter, although **cppy** seems to get a bit more out of it than the other ones.

### Finally

Every suggestions, comments or advices are welcome. Feel free to suggest anything that could make this project even better, the goal is to be able to provide good bindings for libpointmatcher. Thanks!

pomerlef, 2020-06-29:
Thanks for the thorough analysis.  I guess we can start with AutoWIG. 

@simonpierredeschenes any concerns?

wlav, 2020-06-29:
Just adding a couple of remarks; feel free to ignore. :) Full disclosure: I'm the author of cppyy, and saw some of my documents referenced above, which is how I landed here.

First: "all of the above tools have their strenghs and weaknesses". This is an extremely important point as you have "Performance" on the top of your list. Performance in CPython (and PyPy even more so), from an automatic binder, depends on specializing the paths that you care about. If that is already happening out-of-the-box, then you're golden. If not, you're in for some work. I'm afraid that testing and profiling is the only way to find out.

On that point, yes, automatic binders do help a lot with maintenance, but even though they get you 99% of the way, that last bit that they lack is also the hardest part. I.e. compared to a full manual solution, that 1% would be more like 20-30% of the manual effort. Still a gain for sure, but you may be forced to learn more about the tool than you hoped for.

For Eigen/NumPy: note that Eigen's layout may differ depending on compiler settings (e.g. if you enable vector instructions), so simply copies may be the only portable way in some cases. As for cppyy, one problem with Eigen is that template expressions are intended to be squashed by the C++ compiler. This does not naturally happen in cppyy in the way that wrappers are generated, so sometimes you end up with massive expressions escaping to Python. I'm working on Eigen-specific module, but this is a WIP.

Finally, for PyPy, cppyy there has less functionality than on CPython. It will trivially outperform the other solutions, but it does so because there is builtin support for cppyy in PyPy. The others have to go through the emulated C-API which can be very slow. This builtin cppyy module is being separately maintained and because of the higher demand for CPython support, effort has mainly been going to CPython, so cppyy in PyPy is now lagging.

pomerlef, 2020-06-29:
Thanks a lot @wlav ! Your inputs are much appreciated. We will continue our analysis, but I guess we need to do some quick tests with those tools to understand a bit more what is going on.

AlexandreG87, 2020-06-29:
Yes! Thank you very much @wlav! Your feedback is very much appreciated!