norlab-ulaval/libpointmatcher #223: Cmake Error: No target "nabo"

bishesh, 2017-10-09:
cmake results in an error saying that it cannot find target "nabo". I've manually set the directories of Eigen and boost. It somehow automatically finds the libnabo_DIR (the build directory, and not the install directory).
My system: 
g++ (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
ccmake/cmake version 3.5.1

Below is the result with --debug-output

```
Running with debug output on.
   Entering             /home/bk17/Documents/softwares/libpointmatcher/contrib
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Entering             /home/bk17/Documents/softwares/libpointmatcher/contrib/yaml-cpp-pm
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/contrib/CMakeLists.txt
   Returning to         /home/bk17/Documents/softwares/libpointmatcher/contrib
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/contrib/CMakeLists.txt
   Returning to         /home/bk17/Documents/softwares/libpointmatcher
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- Boost version: 1.58.0
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- Found the following Boost libraries:
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   thread
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   filesystem
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   system
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   program_options
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   date_time
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   chrono
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   atomic
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- Boost version: 1.58.0
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- Found the following Boost libraries:
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   thread
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   filesystem
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   system
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   program_options
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   date_time
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   chrono
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
--   atomic
   Called from: [2]	/usr/share/cmake-3.5/Modules/FindBoost.cmake
                [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- libnabo found, version 1.0.6 (include=/home/bk17/Documents/softwares/libnabo;/home/bk17/Documents/softwares/eigen-eigen-5a0156e40feb;/home/bk17/work/dnlclient/build/boost/install/include libs=$<TARGET_FILE:nabo>;gomp)
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
CMake Warning (dev) at CMakeLists.txt:189 (get_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "yaml-cpp-pm".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- using built-in yaml-cpp, version 0.3.0
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   -- text-based configuration enabled
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- API Documentation (doxygen): disabled
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Entering             /home/bk17/Documents/softwares/libpointmatcher/examples
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Returning to         /home/bk17/Documents/softwares/libpointmatcher
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Entering             /home/bk17/Documents/softwares/libpointmatcher/evaluations
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Returning to         /home/bk17/Documents/softwares/libpointmatcher
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Entering             /home/bk17/Documents/softwares/libpointmatcher/utest
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
   Returning to         /home/bk17/Documents/softwares/libpointmatcher
   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
CMake Warning (dev) at CMakeLists.txt:398 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "pointmatcher".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
CMake Error at CMakeLists.txt:320 (target_link_libraries):
  Error evaluating generator expression:

    $<TARGET_FILE:nabo>

  No target "nabo"

   Called from: [1]	/home/bk17/Documents/softwares/libpointmatcher/CMakeLists.txt
-- Configuring incomplete, errors occurred!

```

pomerlef, 2017-10-09:
@HannesSommer is that related to the issues ongoing on libnabo?

HannesSommer, 2017-10-09:
@pomerlef , I'm note sure, yet. It might be an old (solved) problem preserved on the system. 

@bishesh , could you please wipe the folder `~/.cmake/packages/libnabo/` and try building libpointmatcher again? 

And if that works, and you have some extra time to help us improving the code: to execute `cmake .` in libnabo's build folder and try libpointmatcher again (also starting with a `cmake .`).

bishesh, 2017-10-10:
Thanks. I was wondering how libpointmatcher cmake automatically found the libnabo path as build folder; didn't realize that it saves this info in ~/.cmake/packages ! 
Removing that and setting libnabo path to LIBNABO_INSTALL_DIR/share/libnabo/cmake worked. 

About rebuilding libnabo: I had just built libnabo yesterday after using the changed cmakelists file as from [here](https://github.com/ethz-asl/libnabo/issues/67). 
I can rebuild it, but I am not sure if it would change anything because I would be repeating the exact same steps I did yesterday, right ? I didn't understand exactly what you want to check. 
But I'd be happy to try if I understand properly what we are trying here.

HannesSommer, 2017-10-10:
Thanks for trying and reporting!

I didn't forget about you building nabo yesterday ;). The difference is that you've deleted nabo from the cmake user registry in between (`~/.cmake/.`.). If you try once more we learn whether the problem was from an old entry or whether libnabo is still providing bad ones. They are generated in the cmake config time, I'm not mistaken. 
It is important that you explicitly call cmake first for libnabo and then for libpointmatcher.

bishesh, 2017-10-11:
Yes, libnabo still generates the wrong path that gets picked up by libpointmatcher from ~/.cmake/.. 

I deleted the build folders, install folders and ~/.cmake/ directories. Starting again from libnabo, which installs fine, the same problem that I reported in this issue comes. The  problem is with the variable "libnabo_DIR" which gets set to the libnabo build folder by default. Once I manually change it to the cmake directory in the installed folder, things are fine.

HannesSommer, 2017-10-11:
Okay, then we have to fix that. Big thanks for testing and reporting!

Technically it is fine that it finds the build space, because it is supposed to work without install. The issue is with the target then, as the actual error indicates.

closed 2018-05-24 by pomerlef