norlab-ulaval/libpointmatcher #226: libpointmatch seems to round data and lose precision?

nickponline, 2017-10-16:
I registered a pointcloud to itself and it appears to round the x and y values to the nearest whole number:

```
==> cloud.csv <==
x, y, z, R, G, B
539719.26,4065740.41,1.96,38807,34181,30583
539720.41,4065739.15,1.89,38036,33410,29555
539721.55,4065739.18,1.86,40092,34952,31354
539718.28,4065739.35,1.79,41120,36494,32896
539719.41,4065739.17,1.91,42662,38036,34181
```

```
==> cloud_registered.csv <==
539719 , 4.06574e+06 , 1.96032 , 38807,34181,30583
539720 , 4.06574e+06 , 1.89032 , 38036,33410,29555
539722 , 4.06574e+06 , 1.86033 , 40092,34952,31354
539718 , 4.06574e+06 , 1.79032 , 41120,36494,32896
```

Since my data is in meter the rounding is significant. 

Nick

pomerlef, 2017-10-16:
The library can work with `float` or `double`. The executable `pmicp` used `float ` by default.

You can change [this line](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/icp.cpp#L48)  for:
```
typedef PointMatcher<double> PM;
```

nickponline, 2017-10-16:
Do you think that is actually the issue, that's should cause the output value to be rounded to whole numbers?

pomerlef, 2017-10-17:
From a C++ tutorial:

> When outputting floating point numbers, std::cout has a default precision of 6 -- that is, it assumes all floating point variables are only significant to 6 digits, and hence it will truncate anything after that.

This would explain `539719.26` converted to `539719` and clearly `4065740.41` to `4.06574e+06`.

Computing with floating point is faster than double precision, but I saw your problem when working with GPS coordinates. If you want to keep the computational speed, you can express your point cloud in a temporary frame that subtracts the repetitive information.

Did you try to change the template to `double`?

nickponline, 2017-10-17:
Yeah I did try changing float to double. I will try centering the data by
subtracting the mean. In general is that a good practice?

pomerlef, 2017-10-17:
That's what is happening inside libpm anyway to properly resolve the rotations. So, there should be no harm in doing so to preserve the floating precision at the right place for meters.

closed 2018-09-10 by simonpierredeschenes