norlab-ulaval/libpointmatcher #239: Translation not accurate

ashnarayan13, 2018-02-14:
Hi,

I wanted to know if there are translations / rotations limitations for the library. 

Also, are there any limits to the minimum number of points required for the ICP to work?

[image]

I am trying to run ICP on this. The points are rotated by 90° and translated as well. 
After running the libpointmatcher, my results are 

[image]

I am not sure what error is there for the translation as the box was rotated correctly after the ICP. 

My YAML
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.8
referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 3
      ratio: 0.9
matcher:
  KDTreeMatcher:
    knn: 3
    epsilon: 0 
outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.95
errorMinimizer:
  PointToPlaneErrorMinimizer
transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 200
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.01
      minDiffTransErr: 0.01
      smoothLength: 4

When I run the same configuration on data without the 90° rotation, i get a correct translation. 

[image]

Any suggestions? 

Thank you

pomerlef, 2018-02-14:
We are adventuring in the application dependant support, which I try to avoid.

But, one of the main misconception I need to point out here, is that ICP is a local solution not a global one. The points need to be _roughly_ aligned (often manually or using another algorithm) for ICP to converge. The definition of _roughly_ will depend on your application and the configuration of ICP so it's hard to give exact range. In general for the angle, more than 35 deg. is hard to recover. 

The theoretical minimum number of points is 2 in 2D and 3 in 3D. We might have some other specific validation in the library, but I don't recall them specifically.

ashnarayan13, 2018-02-14:
Thank you for the quick response. 

I was expecting the ICP to fail becuase of the rotation, but I am puzzled as to why it handled the rotation and failed to perform the translation correctly. 

That's why I wanted to verify if there is a bug in my setup which is causing the translation to go incorrect.

pomerlef, 2018-02-14:
You can always activate the `VTKFileInspector` to see how the algorithm behaves during the iterative process.

More info here: https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ICPIntro.md

closed 2018-03-07 by ashnarayan13