norlab-ulaval/libpointmatcher #245: MaxPointCount Data Filter may not result as intended

MathLabu, 2018-05-22:
In the implementation of the `MaxPointCountDataPointsFilter` some cast may result in an unwanted value, in particular for the line `index += N * static_cast<unsigned> (static_cast<float>(rand () / double (RAND_MAX)));` (_see complete code below for context_), where the unsigned cast always (_except for at max 1 case over 32767_) result to `0` (as illustrated by [this minimal code (Coliru)](http://coliru.stacked-crooked.com/a/f2a39cf82ad70b73)). However if we put the unsigned cast before the calculation, the `index` may overflow the upper bound `N`. I'm not sure what the code intend to do. 

Can someone indicate me if it's intentional or an error? And if so, how to correct it?

```cpp
/ In-place filter
template<typename T>
void DataPointsFiltersImpl<T>::MaxPointCountDataPointsFilter::inPlaceFilter(
	DataPoints& cloud)
{
	unsigned N = static_cast<unsigned> (cloud.features.cols());
	if (maxCount < N) {
		DataPoints cloud_filtered = cloud.createSimilarEmpty(maxCount);
		std::srand(seed);

		unsigned top = N - maxCount;
		unsigned i = 0;
		unsigned index = 0;
		for (size_t n = maxCount; n >= 2; n--)
		{
			float V = static_cast<float>(rand () / double (RAND_MAX));
			unsigned S = 0;
			float quot = static_cast<float> (top) / static_cast<float> (N);
			while (quot > V)
			{
				S++;
				top--;
				N--;
				quot = quot * static_cast<float> (top) / static_cast<float> (N);
			}
			index += S;
			cloud_filtered.setColFrom(i++, cloud, index++);
			N--;
		}

		index += N * static_cast<unsigned> (static_cast<float>(rand () / double (RAND_MAX)));
		cloud_filtered.setColFrom(i++, cloud, index++);
		PointMatcher<T>::swapDataPoints(cloud, cloud_filtered);
		cloud.conservativeResize(i);
	}
}
```

pomerlef, 2018-05-22:
The more I'm reading it the more I'm wondering how this is implemented. The goal of that fonction is to do random sampling only if the number of points is higher than a given threshold. If so, the random sampling should arrive exactly at `maxCount`.

The other problem is 
```
DataPoints cloud_filtered = cloud.createSimilarEmpty(maxCount);
```
, which do not fit with the idea of in-place filtering.

MathLabu, 2018-05-22:
Okay, I'll propose a fix for this filter after the global refactoring of the data filters. 

Probably something like building a random sequence of `maxCount` elements, pushing iteratively the corresponding column from the beginning of the point cloud and finally resize the point cloud. This may not strictly be random but I think it's a better and quicker alternative.