norlab-ulaval/libpointmatcher #253: Deprecated use of Voxel in Gestalt filter

MathLabu, 2018-05-31:
As the use of voxel should be deprecated, this filter should use an Octree (for 3D points) or a Quadtree (for 2D points) instead of voxels. 

Plus by construction of the octree, the algorithm could be simplified since we have the information of the k-NN in the node. For instance, if we want _k_ points to compute covariance, we just need to build the octree/quadtree with this number of point as criterion (plus, the cells containing less than _k_ points could be ignored, acting like a filter for very sparse space partition).

YoshuaNava, 2021-04-22:
@MathLabu in the second paragraph of your comment above, were you actually suggesting to use the octree for finding neighbors, instead of running K-NN?

For that sake, would you build the octree, and then to query the neighbors of a point, just navigate the octree nodes recursively until reaching the area around that point?

MathLabu, 2021-04-23:
I don't exactly remember what I meant at this time :sweat_smile: 
But I think my comment only applies to the Gestalt filter, so not really suggesting to use the octree instead of a knn for finding neighbors in general.

In this filter, around [here](https://github.com/ethz-asl/libpointmatcher/blob/76f99fce0fe69e6384102a0343fdf8d262626e1f/pointmatcher/DataPointsFilters/Gestalt.cpp#L359), a search for nearest-neighbors is done. 

I guess what I meant was to use the octree as spatial representation instead of the voxel (because of memory issue), and try to adapt the algorithm to leverage the octree for the computation, by applying the a _visitor_ that would perform the actual computation at each leaf (for instance, computing covariance for leaf having at least _k_ points)!

YoshuaNava, 2021-04-23:
@MathLabu I think your idea is quite solid, and applicable to other filters that depend on the neighborhood of points, such as the surface normal (for computing normals based on local neighbors), feature computation, statistical outlier rejection, region growing, etc.

I hope in the future we have the chance to test this.