norlab-ulaval/libpointmatcher #260: DataFilterTest.NormalSpaceDataPointsFilter fails assert in debug mode

crisluengo, 2018-06-25:
[`NormalSpaceDataPointsFilter<T>::inPlaceFilter` has an assert](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/DataPointsFilters/NormalSpace.cpp#L91) to ensure that normals are normalized:

    assert(normals.col(i).head(3).norm() == 1);

This one triggers an exception for me when running [the `DataFilterTest.NormalSpaceDataPointsFilter` test](https://github.com/ethz-asl/libpointmatcher/blob/master/utest/ui/DataFilters.cpp#L490):

> utest: /home/cris/SW/SOUP/libpointmatcher/pointmatcher/DataPointsFilters/NormalSpace.cpp:91: void NormalSpaceDataPointsFilter<T>::inPlaceFilter(NormalSpaceDataPointsFilter<T>::DataPoints&) [with T = float; NormalSpaceDataPointsFilter<T>::DataPoints = PointMatcher<float>::DataPoints]: Assertion `normals.col(i).head(3).norm() == 1' failed.

I replaced the assert with an `if() std::cout << ...` statement, which results in a long list of normals being displayed. They are all in the range `[1,sqrt(2)]`, which leads me to believe that these are 2D vectors that have been normalized such that the max element is 1 (L_infinity norm).

These all happen when `nbSample` is 240 (the `2*nbPts2D/3` case in the test), not for other values of `nbSample`.

All tests (including this one with the `assert` disabled) pass successfully.

pomerlef, 2018-06-26:
@MathLabu could you have a look?

MathLabu, 2018-06-26:
Hi!

In fact the `NormalSpaceDataPointsFilter` was not designed to handle 2D point-cloud. I mechanically added the 2D test case (this is where the assertion failed) in the `utest`, but that was a mistake.

About the assert, we need normalized-normals to compute the _normal-space_ as a 2D space in polar coordinates _Phi-Theta_ (_Rho_=1, so no need to take it into account).

I propose 2 solutions to fix this issue:
- add support for 2D case in the filter (just need to build _normal-space_ in 2D, i.e. a polar angle histogram)
- remove the 2D test case

Do you need the 2D support? If so I can implement it quite quickly, otherwise just removing the 2D test case should be enough.

crisluengo, 2018-06-26:
Thanks for your insight. Makes me feel better about having removed that test. :)

When `nbSample=nbPts2D`, is it not also a 2D test case? That one doesn't trigger the assertion.

MathLabu, 2018-06-26:
You're welcome ! I'm going to do a PR to fix the issue adding a check of point cloud dimension in the filter.

`nbSample=nbPts2D` is a 2D test, but this one does not trigger the assertion because there is a [check](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/DataPointsFilters/NormalSpace.cpp#L69) at the beginning of the filter to verify if the number of sample requested is superior or equal to the number of points: if so, nothing is done, otherwise we perform the algorithm.

PR #261 (2018-06-26): Quick fix for 2D test case in NormalSpace filter
(fixes #260)

As addressed by #260, the `NormalSpaceDataPointsFilter` trigger an assertion in debug mode due to unsupported 2D point cloud.

Temporary marked the filter as not supporting the 2D case:
- add check for point cloud dimension in filter
- remove 2D test case

**TODO:** add 2D support by building an histogram of 2D normals and uniformly sampling in it.
closed 2018-06-26 by pomerlef