norlab-ulaval/libpointmatcher #262: first time comiplation error

MahdiSMIDA, 2018-07-02:
OctreeGrid.cpp:295:42: error: cannot call member function ‘S PointMatcherSupport::Parametrizable::get(const string&) [with S = int; std::__cxx11::string = std::__cxx11::basic_string<char>]’ without object
   const int sm = Parametrizable::get<int>("samplingMethod");

Name            | Version
----------------|-------------------------------
ubuntu:         | Description: Ubuntu 17.10
architecture:   | 64-bit
gcc:            |
git:            | git version 2.14.1
cmake:          | cmake version 3.9.1 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost:          | Version: 1.62.0.1
eigen3:         | Version: 3.3.4-3
doxygen:        | Version: 1.8.13-9

pomerlef, 2018-07-02:
@MathLabu Could you have a look? It seems to be from `OctreeGrid.cpp`.

MathLabu, 2018-07-03:
Hi!

I can't see anything wrong with this [line](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/DataPointsFilters/OctreeGrid.cpp#L295): as we only call the super method [`get`](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/Parametrizable.h#L173) of the `Parametrizable` class, you should not have the: 
> error: cannot call member function "..." without object

kind of error, since `OctreeGridDataPointsFilter<T>` inherits of `PointMatcher<T>::DataPointsFilter` which inherits of `Parametrizable` (see [here](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/PointMatcher.h#L433))... 

As you are on Ubuntu 17.10, I suppose your gcc version is something like 7.x? If there is something wrong with this call, you should have an error [here](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/DataPointsFilters/MaxPointCount.cpp#L47) too since we call the same method the same way.

Maybe if you provide more information I could find if there is something wrong, but with just this error, I can't see anything wrong, sorry!

tobiasneumann, 2018-07-04:
I can reproduce this with Fedora 27 and gcc 7.3.1.

I also don't understand how that could work without an instance.
Method `get` calls `getParamValueString` which uses the `parameters` and `parametersUsed` member.

However when I remove code lines 293-299, 301 in `OctreeGrid.cpp` and 53-59 and 61 in `CovarianceSampling.cpp` I can compile so that means the method call `Parametrizable::get<T>(std::string)` in lines 289, 290, 291 and so on does work for some reason, *confusing*.

closed 2018-07-31 by pomerlef