norlab-ulaval/libpointmatcher #264: Error when use cmake to build libnabo on win10

arcosintan, 2018-07-12:
Hi, I meet some issue when I use cmake to compile this on win10, this is the error message:

> (Link target) ->
> 
> invalid_matrix_types.obj : **error** LNK2019: unresolved external symbol "public: static struct Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> > * __cdecl Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::create(class Eigen::Matrix<float,-1,-1,0,-1,-1> const &,int,enum Nabo::NearestNeighbourSearch<float,class Eigen::Matrix<float,-1,-1,0,-1,-1> >::SearchType,unsigned int,struct Nabo::Parameters const &)" (?create@?$NearestNeighbourSearch@MV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@@Nabo@@SAPEAU12@AEBV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@HW4SearchType@12@IAEBUParameters@2@@Z) referenced in function "int __cdecl testFunction<class Eigen::Matrix<float,-1,-1,0,-1,-1> >(void)" (??$testFunction@V?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@@@YAHXZ) [C:\libnabo\build_libnabo_vs2017_x64\tests\CMakeFiles\CMakeTmp\cmTC_b6580.vcxproj]
> 
> C:\libnabo\build_libnabo_vs2017_x64\tests\CMakeFiles\CMakeTmp\Debug\cmTC_b6580.exe : fatal **error** LNK1120: 1 unresolved externals [C:\libnabo\build_libnabo_vs2017_x64\tests\CMakeFiles\CMakeTmp\cmTC_b6580.vcxproj]
> 
> 
> 
> 
> 
>   1 Warning(s)
> 
>   2 Error(s)

And this is my cmake setting:

[image]

cao-chao-telecom-sudparis, 2018-08-23:
Hi, which version of libnabo are you using? Did you try using another version of Cmake? I remembered that I used to use 3.12.0 and have the same error then I switched to 3.2.3 and it worked. Seems like the latest Cmake have some compatible issues with libnabo and boost sometimes.

simonpierredeschenes, 2018-09-10:
This problem comes from libnabo. Until this problem is fixed, a temporary solution is to remove the line ``add_subdirectory(tests)`` from the main CMakeLists.txt of libnabo.

closed 2018-09-17 by simonpierredeschenes