norlab-ulaval/libpointmatcher #265: 2D matching pipe line

KrzysztofZurad, 2018-07-17:
I've been struggling a bit to make libpointmatcher work with 2D case. My current setup:

I have a 3D point cloud projected to 2D (z axis = 0 for every point, because of specificity of my application a 2D projection seems reasonable).
I would like to calculate normal vectors and use shadow filter to eliminate outliers, however it seems to be impossible. Setting up z axis of each point to 0 gives me normal vectors all pointing up.
I would like to achieve normal vector calculation only for 2D case. I've tried several solutions, without sucess.

I've used a pmicp executable **with the same reference and reading point clouds**, for the sake of simplicity.

Option 1:
I've added an extra DataPointsFilter that removes 3'rd dimension from the data structure, however when I visualize output in ParaView I get such results:

for reference cloud:
[image]

Which seems to be all fine, however for the reading cloud I get:

[image]

As you can see it has been transformed in very weird way, that I don't understand at all.

Option 2:
I've manually removed z axis point from the input file, leaving only x and y component. I got

reference cloud:
[image]

reading cloud:
[image]

My pipeline config files has been setup to:

```yaml
readingDataPointsFilters:
 - ProjectOrthoDataPointsFilter
 - SimpleSensorNoiseDataPointsFilter
 - SurfaceNormalDataPointsFilter:
     knn: 5
     keepNormals: 1
     keepDensities: 1
 #- MaxDensityDataPointsFilter:
 #    maxDensity: 10000
 - ObservationDirectionDataPointsFilter
 - OrientNormalsDataPointsFilter

referenceDataPointsFilters:
 - ProjectOrthoDataPointsFilter
 - SimpleSensorNoiseDataPointsFilter
 - SurfaceNormalDataPointsFilter:
     knn: 5
     keepNormals: 1
     keepDensities: 1
 #- MaxDensityDataPointsFilter:
 #    maxDensity: 10000
 - ObservationDirectionDataPointsFilter
 - OrientNormalsDataPointsFilter

matcher:
 KDTreeMatcher

outlierFilters:
   - NullOutlierFilter

errorMinimizer:
   IdentityErrorMinimizer

transformationCheckers:
   - CounterTransformationChecker

inspector:
  VTKFileInspector:
     baseFileName : vissteps
     dumpDataLinks : 1
     dumpReading : 1
     dumpReference : 1

logger:
  FileLogger

```
where ProjectOrthoDataPointsFilter is filter that I have tried to implement that simply removes a 3rd dimention from the point cloud.

```cpp
// ProjectOrthoDataPointsFilter
// Constructor
template<typename T>
DataPointsFiltersImpl<T>::ProjectOrthoDataPointsFilter::ProjectOrthoDataPointsFilter(const Parameters& params):
	DataPointsFilter("ProjectOrthoDataPointsFilter", ProjectOrthoDataPointsFilter::availableParameters(), params),
	zPlane(Parametrizable::get<T>("zPlane"))
{
    LOG_INFO_STREAM("Using ProjectOrthoDataPointsFilter with zPlane=" << zPlane);
}

// Compute
template<typename T>
typename PointMatcher<T>::DataPoints DataPointsFiltersImpl<T>::ProjectOrthoDataPointsFilter::filter(
	const DataPoints& input)
{
	DataPoints output(input);
	inPlaceFilter(output);
	return output;
}

// In-place filter
template<typename T>
void DataPointsFiltersImpl<T>::ProjectOrthoDataPointsFilter::inPlaceFilter(
	DataPoints& cloud)
{
    const int nbPointsIn = cloud.features.cols();
    typedef PointMatcher<T> PM;
    typedef typename PM::DataPoints DataPoints;
    typedef typename DataPoints::Label Label;
    typedef typename DataPoints::Labels Labels;

    Labels featLabels;
    featLabels.push_back(Label("x", 1));
    featLabels.push_back(Label("y", 1));
    featLabels.push_back(Label("pad", 1));

    Labels descLabels;

    DataPoints cloud_filtered(featLabels, descLabels, nbPointsIn);
    // cloud_filtered.getFeatureViewByName("pad")

    int j = 0;
    for (int i = 0; i < nbPointsIn; i++)
    {
        cloud_filtered.features(0, i) = cloud.features.col(i).array()[0];
        cloud_filtered.features(1, i) = cloud.features.col(i).array()[1];
        cloud_filtered.features(2, i) = 1;
        j++;
    }

    // cloud = cloud_filtered;
    PointMatcher<T>::swapDataPoints(cloud, cloud_filtered);

    // for (int i = 0; i < nbPointsIn; i++)
    // {
    //     std::cout << "Cloud " << cloud.features.col(i).array() << std::endl;
    //     std::cout << "Cloud_filtered " << cloud_filtered.features.col(i).array() << std::endl;
    // }

    // cloud.conservativeResize(j);
}

template struct DataPointsFiltersImpl<float>::ProjectOrthoDataPointsFilter;
template struct DataPointsFiltersImpl<double>::ProjectOrthoDataPointsFilter;
```

am I missing something here? How should I setup a 2D processing pipeline to correctly filter and scan match my input?

Since I couldn't find many examples of libpointmatcher in 2D, I've been wondering what should be the proper processing pipeline setup.

pomerlef, 2018-07-19:
Hi @KrzysztofZurad ,

If you have 2D point clouds, you should setup the pipeline in 2D. So your second solution is the way to go.

The surface normals are extracted using an eigendecomposition. The normal vector is selected using the smallest eigenvalue. If of all z are set to zero, this will trigger the smallest eigenvalue and the normal will follow the z-axis.

KrzysztofZurad, 2018-07-19:
Thanks for you prompted response.
After realizing that I can't create a filter that will remove 3rd dimension from the input pointcloud due to fact that a size of transformation matrix is set in the constructor of icp algorithm and cannot be changed during runtime, i've managed to solve my issue.
I have added a Force2D parameter in libpointmatcher_ros package to get rid of the 3rd dimension.

```c++

typename PointMatcher<T>::DataPoints rosMsgToPointMatcherCloud(const sensor_msgs::PointCloud2& rosMsg, const bool force2D)

if (name == "x" || name == "y" || (name == "z" && !force2D))
{
	featLabels.push_back(Label(name, count));
	isFeature.push_back(true);
}
```

Now it works fine.
Shall I make a pull request and add this as a feature to libpointmatcher_ros??

closed 2018-07-31 by pomerlef

gjcliff, 2024-10-28:
For anyone looking to use libpointmatcher for 2d point clouds, I found an easier way to accomplish this in one of the example files that doesn't require modifying and building the repo from source. It's [here](https://github.com/norlab-ulaval/libpointmatcher/blob/master/examples/icp_advance_api.cpp)