norlab-ulaval/libpointmatcher #266: DeprecationWarning.h missing after install

davidlandry93, 2018-07-19:
Hello all,

I think that a bug sneaked into the installation procedure.
When I build and install libpointmatcher and then try to include it from a consumer project I get

```
/home/dlandry/local/include/pointmatcher/PointMatcher.h:62:33: fatal error: DeprecationWarnings.h: No such file or directory
```

I think this might have something to do with change @lorenwel made in #210, probably just forgot to add DeprecationWarnings.h to the CMakeLists.txt.

Can anybody reproduce?

Thanks

HannesSommer, 2018-07-20:
Thanks, @davidlandry93 , for pointing this out! Yes, this is missing. No need to reproduce since it must be installed by principle as a transitive dependency of `PointMatcher.h`. 
It should be around here https://github.com/ethz-asl/libpointmatcher/blob/master/CMakeLists.txt#L374.

mpitropov, 2018-08-14:
@HannesSommer Would you be able to review my PR? https://github.com/ethz-asl/libpointmatcher/pull/274 The program is currently not installed correctly.

PR #274 (2018-08-21): Added DeprecationWarnings.h to install
Closes #266 

Added missing install in order to fix this error:

```
Errors     << libpointmatcher_ros:make /home/wavelab/catkin_ws/logs/libpointmatcher_ros/build.make.002.log
In file included from /home/wavelab/catkin_ws/src/ethzasl_icp_mapping/libpointmatcher_ros/include/pointmatcher_ros/point_cloud.h:4:0,
                 from /home/wavelab/catkin_ws/src/ethzasl_icp_mapping/libpointmatcher_ros/src/point_cloud.cpp:1:
/usr/local/include/pointmatcher/PointMatcher.h:62:33: fatal error: DeprecationWarnings.h: No such file or directory
compilation terminated [... cut]
closed 2018-08-21 by pomerlef