norlab-ulaval/libpointmatcher #277: Add clang format

PhiBabin, 2018-09-26:
It would be easier to submit pull request, if the indentation/format rules were enforced by an automatic tools like [clang-format](https://clang.llvm.org/docs/ClangFormat.html).

pomerlef, 2018-09-26:
Sounds like a good idea.

simonpierredeschenes, 2018-10-15:
I don't know if this is what we want. When we will add the formater, the first commit will contain a huge amount of lines changed, which will ruin git blame in almost all files. This must be taken into account before adding it.

simonpierredeschenes, 2018-10-15:
I think it would more reasonable, since history is less important between major releases.

stephanemagnenat, 2018-10-16:
A note about that. We had a similar discussion in [aseba](https://github.com/aseba-community/aseba). Beside breaking the history, automatic formatting also breaks some special formatting done on purpose by the programmer (for example to define a matrix or highlight the specifics of an equation). For new code, a `// clang-format off/on` comment allows to disable automatic formatting, and thus it is acceptable. But for old code, I do not see the benefit. My advise is to apply it to new and changed code only.

pomerlef, 2018-10-16:
Specifications:
- [ ]  Encode our loosely defined formatting rules
- [ ]  Limit automatic parsing to PR
- [ ]  Add delimiters to protect from automatic formatting, this could be `// clang-format off/on`

stephanemagnenat, 2018-10-16:
[This file](https://github.com/aseba-community/aseba/blob/master/.clang-format) might be a useful starting point for the formatting rules, as both aseba and libpointmatcher kind of follow "my" style.

PhiBabin, 2018-10-16:
I try 2 weeks ago to create a clang-format which cover the [libpointmatcher style](https://gist.github.com/PhiBabin/f75d35ce3bac27899f2a0b5770c0e880). It seems libpointmatcher follows aseba style.
We could add a git commit hook to the project, which run clang format on the modified files when creating a commit.
I suggest to add this feature after #282 is merged, since `boost::list_of`are not correctly handle by clang-format.

YoshuaNava, 2020-02-06:
Hi all,

Would it be possible to re-consider integrating a formatter? Both to old and new code. It would simplify debugging, reviewing pull requests, and contributing (if you have a fork), without spending too much time on style discussions and fixing syntax.

pomerlef, 2020-02-06:
I fine with the idea. Could you prepare a PR?

YoshuaNava, 2020-04-23:
Yes.

I propose using clang format, which is easy to run with modern IDEs (for example, VSCode), is also used by other ASL projects, and gives quite a bit of flexibility.

I would divide the contribution in two PRs

* The actual formatting tool, along with a small example on how to run it.
* Formatting the repo or some files at a time.