norlab-ulaval/libpointmatcher #280: How to compute the relative position

MHarbi, 2018-10-01:
I have consecutive point clouds. Their origin is (0, 0, 0). I am looking for finding the relative position, which is composed of (delta_x, delta_y, delta_z). 
I have generated the transformation matrix between each reference and source point clouds, but I could not figure out how to compute the relative position.

Does this library helps me to find this position? How?

simonpierredeschenes, 2018-10-08:
The relative position (delta_x, delta_y, delta_z) between two point clouds corresponds to the translation described by the transformation matrix.

pomerlef, 2018-10-10:
We can add a small explanation on rigid transformation in https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Pointclouds.md