norlab-ulaval/libpointmatcher #283: Compile error with MatchersImpl

durka, 2018-10-05:
Trying to compile the latest master I get several errors like this:

```
[ 51%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/InspectorsImpl.cpp.o
/tmp/libpointmatcher/pointmatcher/MatchersImpl.cpp: In instantiation of ‘typename PointMatcher<T>::Matches MatchersImpl<T>::KDTreeMatcher::findClosests(const DataPoints&) [with T = float; typename PointMatcher<T>::Matches = PointMatcher<float>::Matches; MatchersImpl<T>::DataPoints = PointMatcher<float>::DataPoints]’:
/tmp/libpointmatcher/pointmatcher/MatchersImpl.cpp:103:38:   required from here
/tmp/libpointmatcher/pointmatcher/MatchersImpl.cpp:96:34: error: ‘InvalidIndex’ is not a member of ‘MatchersImpl<float>::NNS {aka Nabo::NearestNeighbourSearch<float>}’
  static_assert(NNS::InvalidIndex == Matches::InvalidId, "");
                ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/tmp/libpointmatcher/pointmatcher/MatchersImpl.cpp:97:34: error: ‘InvalidValue’ is not a member of ‘MatchersImpl<float>::NNS {aka Nabo::NearestNeighbourSearch<float>}’
  static_assert(NNS::InvalidValue == Matches::InvalidDist, "");
                ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
```

pomerlef, 2018-10-05:
To help us a bit, please give us more details about your system using the script `./utest/listVersionsUbuntu.sh` or follow the instructions here: https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Compilation.md#having-problems

From what I see, it seems to be related to `libnabo`.

durka, 2018-10-08:
Here's what I get from that script:

```
Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 18.04.1 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
git: 		| git version 2.17.1
cmake: 		| cmake version 3.10.2 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: 		| Version: 1.65.1.0ubuntu1
eigen3: 	| Version: 3.3.4-4
doxygen: 	| Version: 1.8.13-10
```

simonpierredeschenes, 2018-10-08:
Are you sure that libnabo is up to date and that it was compiled and installed correctly? My computer has exactly the same configuration as yours and I cannot reproduce this error.

pomerlef, 2018-10-10:
Same problem here since 2017: http://build.ros.org/job/Idev__libpointmatcher__ubuntu_trusty_amd64/

`libnabo found, version 1.0.6 (include=/opt/ros/indigo/include libs=/opt/ros/indigo/lib/libnabo.a;gomp)`

HannesSommer, 2018-10-11:
IMO the problem is that this job on build.ros.org is compiling libpointmatcher master together with the most recent release of libnabo (1.0.6), which are incompatible, as @simonpierredeschenes  observed correctly.
The rational behind the ros build farm behavior is probably that a master should compile with the most recent releases of its dependencies, which libpointmatcher's master violates.
Of course libnabo should have gotten a new release long ago...

In any case, the simplest solution to this issue should be to compile libpointmatcher master with libnabo master.

closed 2018-10-26 by simonpierredeschenes