norlab-ulaval/libpointmatcher #290: release 1.2.4 ICP may produce wrong  tranform

lingerer, 2018-10-24:
I use pointmatcher for 2d laser mapping.The new 1.2.4 release may produce some wrong transform matrix in 0,0 or 0,1 when using icp. The value for sin or cos is over 1 or -1.
I ever modified 1.2.3 for eigen 3.3 and never encount this problem, so I guess it must be something wrong for the newly udpate.

pomerlef, 2018-10-24:
Thanks for reporting this problem rapidly. 

Could you give us enough details to replicate the bug on our side?

lingerer, 2018-10-25:
I re-test with multi config and quite sure this is my misunderstanding.
I use PointToPointSimilarityErrorMinimizer for test and the old code treat the matrix as a ragid transform matix as before. I set the config back to pointtopoint or pointplane will not happen.
So you can close this issue now.

closed 2018-10-25 by lingerer

lingerer, 2018-10-30:
I think it should add doc for PointToPointSimilarityErrorMinimizer

simonpierredeschenes, 2018-10-30:
Thank you for you suggestion! We will try to address it as soon as possible. However, please create a new issue when the subject differs. It will help us to keep track of issues.