norlab-ulaval/libpointmatcher #298: Error compiling on ROS kinetic

aaravrav142, 2018-11-06:
Hi, 

I am having the following errors when compiling on ROS Kinetic. 
I have catkin_make_isolated the libnabo and libpointmatcher. 
```
error: no matching function for call to ‘std::unique_ptr<PointMatcher<float>::Transformation>::unique_ptr(std::shared_ptr<PointMatcher<float>::Transformation>)’
  trailCount(0)
```

I would appreciate any help. Thanks.

pomerlef, 2018-11-06:
I suggest you to compile and install libnabo and libpointmacher from master and without `catkin_make_isolated`.

Just follow the regular instructions here: https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Compilation.md

aaravrav142, 2018-11-09:
Hi, 
Thanks. Compiling and building outside the catkin_make_isolated resolved the issue. The package ethzasl_icp_mapper build as well. 
I am now having another issue. When launching the velodyne_puck launch file to test with the velodyne rosbag. Its throwing the following error

`ERROR: cannot launch node of type [ethzasl_icp_mapper/dynamic_mapper]: can't locate node [dynamic_mapper] in package [ethzasl_icp_mapper]`

I can however see the dynamic_mapper.cpp inside the ethzasl_icp_mapper/src. Any hints? 
Thanks

gawela, 2018-11-09:
Did you source your catkin workspace before attempting to launch the node? I.e. 
`source ~/catkin_ws/devel/setup.bash`

aaravrav142, 2018-11-12:
I met with another issue after cleaning the catkin workspace. The error is related to shared pointer from the dynamic_mapper node. Not sure how to resolve this - 
`home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp: In constructor ‘MapMaintener::MapMaintener(ros::NodeHandle&, ros::NodeHandle&)’:
/home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp:136:47: error: no matching function for call to ‘std::unique_ptr<PointMatcher<float>::Transformation>::unique_ptr(std::shared_ptr<PointMatcher<float>::Transformation>)’
  mapFrame(getParam<string>("map_frame", "map"))
                                               ^
In file included from /usr/include/c++/5/bits/locale_conv.h:41:0,
                 from /usr/include/c++/5/locale:43,
                 from /usr/include/c++/5/iomanip:43,
                 from /usr/include/boost/math/policies/error_handling.hpp:12,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp:5:
/usr/include/c++/5/bits/unique_ptr.h:228:2: note: candidate: template<class _Up, class> std::unique_ptr<_Tp, _Dp>::unique_ptr(std::auto_ptr<_Up>&&)
  unique_ptr(auto_ptr<_Up>&& __u) noexcept;
  ^
/usr/include/c++/5/bits/unique_ptr.h:228:2: note:   template argument deduction/substitution failed:
/home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp:136:47: note:   ‘std::shared_ptr<PointMatcher<float>::Transformation>’ is not derived from ‘std::auto_ptr<Y>’
  mapFrame(getParam<string>("map_frame", "map"))
                                               ^
In file included from /usr/include/c++/5/bits/locale_conv.h:41:0,
                 from /usr/include/c++/5/locale:43,
                 from /usr/include/c++/5/iomanip:43,
                 from /usr/include/boost/math/policies/error_handling.hpp:12,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp:5:
/usr/include/c++/5/bits/unique_ptr.h:220:2: note: candidate: template<class _Up, class _Ep, class> std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Up, _Ep>&&)
  unique_ptr(unique_ptr<_Up, _Ep>&& __u) noexcept
  ^
/usr/include/c++/5/bits/unique_ptr.h:220:2: note:   template argument deduction/substitution failed:
/home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp:136:47: note:   ‘std::shared_ptr<PointMatcher<float>::Transformation>’ is not derived from ‘std::unique_ptr<_Tp, _Dp>’
  mapFrame(getParam<string>("map_frame", "map"))
                                               ^
In file included from /usr/include/c++/5/bits/locale_conv.h:41:0,
                 from /usr/include/c++/5/locale:43,
                 from /usr/include/c++/5/iomanip:43,
                 from /usr/include/boost/math/policies/error_handling.hpp:12,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/arav/catkin_ws/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/map_maintainer.cpp:5:
/usr/include/c++/5/bits/unique_ptr.h:205:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>&&) [with _Tp = PointMatcher<float>::Transformation; _Dp = std::default_delete<PointMatcher<float>::Transformation>]
       unique_ptr(unique_ptr&& __u) noexcept
       ^
/usr/include/c++/5/bits/unique_ptr.h:205:7: note:   no known conversion for argument 1 from ‘std::shared_ptr<PointMatcher<float>::Transformation>’ to ‘std::unique_ptr<PointMatcher<float>::Transformation>&&’
/usr/include/c++/5/bits/unique_ptr.h:200:17: note: candidate: constexpr std::unique_ptr<_Tp, _Dp>::unique_ptr(std::nullptr_t) [with _Tp = PointMatcher<float>::Transformation; _Dp = std::default_delete<PointMatcher<float>::Transformation>; std::nullptr_t = std::nullptr_t]
       constexpr unique_ptr(nullptr_t) noexcept : unique_ptr() { }
                 ^
/usr/include/c++/5/bits/unique_ptr.h:200:17: note:   no known conversion for argument 1 from ‘std::shared_ptr<PointMatcher<float>::Transformation>’ to ‘std::nullptr_t’
/usr/include/c++/5/bits/unique_ptr.h:193:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>::pointer, typename std::remove_reference<_To>::type&&) [with _Tp = PointMatcher<float>::Transformation; _Dp = std::default_delete<PointMatcher<float>::Transformation>; std::unique_ptr<_Tp, _Dp>::pointer = PointMatcher<float>::Transformation*; typename std::remove_reference<_To>::type = std::default_delete<PointMatcher<float>::Transformation>]
       unique_ptr(pointer __p,
       ^
/usr/include/c++/5/bits/unique_ptr.h:193:7: note:   candidate expects 2 arguments, 1 provided
/usr/include/c++/5/bits/unique_ptr.h:181:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>::pointer, typename std::conditional<std::is_reference<_Dp>::value, _Dp, const _Dp&>::type) [with _Tp = PointMatcher<float>::Transformation; _Dp = std::def [... cut ...]
ethzasl_icp_mapping/ethzasl_icp_mapper/CMakeFiles/map_maintainer.dir/build.make:62: recipe for target 'ethzasl_icp_mapping/ethzasl_icp_mapper/CMakeFiles/map_maintainer.dir/src/map_maintainer.cpp.o' failed
make[2]: *** [ethzasl_icp_mapping/ethzasl_icp_mapper/CMakeFiles/map_maintainer.dir/src/map_maintainer.cpp.o] Error 1
CMakeFiles/Makefile2:41833: recipe for target 'ethzasl_icp_mapping/ethzasl_icp_mapper/CMakeFiles/map_maintainer.dir/all' failed
make[1]: *** [ethzasl_icp_mapping/ethzasl_icp_mapper/CMakeFiles/map_maintainer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`

PhiBabin, 2018-11-12:
Is this related to changes of shared_ptr to unique_ptr at some places in the code?

simonpierredeschenes, 2018-11-12:
This might be related to libpointmatcher release 1.3.0. We are working to fix this problem in ethzasl_icp_mapping.

simonpierredeschenes, 2018-11-12:
A new commit was pushed to etzasl_icp_mapping. If you pull the new changes, the problem should be fixed.

aaravrav142, 2018-11-14:
Hi, 
The latest pull solved the problem. Thanks a lot. 
Closing.

closed 2018-11-14 by aaravrav142