norlab-ulaval/libpointmatcher #309: Is that possible to achieve good ICP alignment in this situation?

bxc237, 2018-12-26:
Hi,
Could anyone help me to check this alignment problem?
The point cloud was captured by a VLP-16 in a big lab.
The white points are the reference points (the whole lab), red points are the input points (just a part of the lab) and green points are the aligned points after performed ICP. As shown in the picture, the aligned points are totally wrong (the correct alignment should be rotating the red points clockwise), ICP even downscaled the input points. The video of the ICP alignment is in this link (https://youtu.be/2qAyTu_a6Sc), please take a look.
[image]
I do not think the input point cloud is too far away from the reference one and the rotation is not that large. The matcher should be capable to correctly aligned those points, right? 
I would be most grateful if anyone might be able to give me some advice. THX!!

bxc237, 2018-12-27:
Hi, @PhiBabin, please go to this link https://www.dropbox.com/sh/9uwnec8jlbc88lo/AACbUgGGottcGodlIHjUmKnpa?dl=0, which includes yaml, read and reference csv, so that you can reproduce this problem in libpointmatcher. Thanks....

PhiBabin, 2018-12-27:
It seems I was mistaken, there is a minimizer that support a minimization with translation+rotation+scale. You are using `PointToPointSimilarityErrorMinimizer` which can scale the point cloud. Since your point cloud don't seem to require rescaling, I would suggest using `PointToPointErrorMinimizer` or even better `PointToPlaneErrorMinimizer`.

simonpierredeschenes, 2019-01-10:
It looks like the problem is that the distance between the reference and the reading is too important. By specifying an initial rotation of -45 degrees about the y axis, I managed to get a good alignment:
```
pmicp --initRotation [0.707,0,-0.707,0,1,0,0.707,0,0.707] --config default.yaml reference.csv read.csv
```

[image]

bxc237, 2019-01-14:
@simonpierredeschenes Thank you. I can also achieve a good result if the presumption of the relative translation between ref and read-in points cloud is given. However, it seems to be impossible to achieve good alignment without the presumption of rotation/translation. I will find a way to align them without the presumption and post here. Thank you both very much! @PhiBabin @simonpierredeschenes

closed 2019-01-14 by bxc237