norlab-ulaval/libpointmatcher #322: Failing to build from source on Indigo after libnabo 1.0.7 release

tfoote, 2019-03-01:
From: http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__libpointmatcher__ubuntu_trusty_amd64__binary/4/console

which was triggered by the libnabo 1.0.7 release that appears to introduce c++11 elements into the header `nabo.h` either libnabo needs to export c++11 or libpointmatcher needs to declare it.

@simonpierredeschenes You appear to be contributing to both packages so I'm filing this here in the hopes that this can be fixed instead of just reverting the libnabo release: https://github.com/ros/rosdistro/pull/20360

```
[ 37%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Logger.cpp.o
/usr/bin/x86_64-linux-gnu-g++   -Dpointmatcher_EXPORTS -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2  -fPIC -I/usr/include/eigen3 -I/opt/ros/indigo/include -I/tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/contrib/yaml-cpp-pm/include -I/tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3    -O3 -Wall -o CMakeFiles/pointmatcher.dir/pointmatcher/Logger.cpp.o -c /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/Logger.cpp
In file included from /usr/include/c++/4.8/type_traits:35:0,
                 from /opt/ros/indigo/include/nabo/third_party/any.hpp:20,
                 from /opt/ros/indigo/include/nabo/nabo.h:44,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/PointMatcher.h:47,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/Logger.cpp:36:
/usr/include/c++/4.8/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
In file included from /opt/ros/indigo/include/nabo/nabo.h:44:0,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/PointMatcher.h:47,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/Logger.cpp:36:
/opt/ros/indigo/include/nabo/third_party/any.hpp:29:5: warning: identifier ‘noexcept’ is a keyword in C++11 [-Wc++0x-compat]
     const char* what() const noexcept override
     ^
/opt/ros/indigo/include/nabo/third_party/any.hpp:40:9: warning: identifier ‘nullptr’ is a keyword in C++11 [-Wc++0x-compat]
         vtable(nullptr)
         ^
In file included from /opt/ros/indigo/include/nabo/nabo.h:44:0,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/PointMatcher.h:47,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/Logger.cpp:36:
/opt/ros/indigo/include/nabo/third_party/any.hpp:79:9: warning: identifier ‘static_assert’ is a keyword in C++11 [-Wc++0x-compat]
         static_assert(std::is_copy_constructible<typename std::decay<ValueType>::type>::value,
         ^
In file included from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/PointMatcher.h:47:0,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/Logger.cpp:36:
/opt/ros/indigo/include/nabo/nabo.h:217:2: warning: identifier ‘constexpr’ is a keyword in C++11 [-Wc++0x-compat]
  inline constexpr IndexType invalidIndex() {
  ^
In file included from /opt/ros/indigo/include/nabo/nabo.h:44:0,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/PointMatcher.h:47,
                 from /tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/pointmatcher/Logger.cpp:36:
/opt/ros/indigo/include/nabo/third_party/any.hpp:29:24: error: expected ‘;’ at end of member declaration
     const char* what() const noexcept override
```

tfoote, 2019-03-01:
This is the only regression in indigo. I'm preparing to [sync](https://discourse.ros.org/t/preparing-for-indigo-sync-2019-03-01/8079?u=tfoote) if we can get a fix in early next week that would be great. Otherwise I plan to roll back the libnabo release, https://github.com/ros/rosdistro/pull/20360

simonpierredeschenes, 2019-03-01:
@tfoote I will do my best to release a working version of libpointmatcher by monday or tuesday.

tfoote, 2019-03-02:
@simonpierredeschenes Great thanks!

tfoote, 2019-03-05:
Thanks this looks to be resolve with build http://build.ros.org/job/Ibin_uT64__libpointmatcher__ubuntu_trusty_amd64__binary/16/

closed 2019-03-05 by tfoote