norlab-ulaval/libpointmatcher #334: namespaces and functions not available when linking to LibPointMathcer with Cmake

coleharlow12, 2019-05-02:
Hi I am interested in building off of the LibPointMatcher to create a boresighting calibration package for a drone. A lot of the functionality I am looking for is provided in the ICP.cpp file in the included examples of LPM. I would like to compile this example in a separate project and then build off of this file. To do this I should just be able to follow the directions to link a project to LPM. When I do this though while the CMakeLists.txt is able to find the LPM package and link it to the executable, it appears that a lot of the functions and namespaces available within the LPM library are not available when LPM is linked to. For example in the ICP.cpp file the pointmatcher/Bibliography.h file is included. When trying to create an executable from ICP.cpp in the external package this library cannot be found. I worked around this by adding pointmatcher/Bibliography.h to the install files in the LibPointMatcher top level CMakelists.txt file then rebuilding and installing LPM. While this fixes the initial problem It leads to a huge number of errors being thrown which make reference to many of the dependencies of LPM. If anyone could explain what I am doing wrong it would greatly be appreciated.

PhiBabin, 2019-05-02:
`ICP.cpp` is not meant to be use outside of the library. Thus, you should not try to create an executable out of `ICP.cpp`. What feature do you actually need from `ICP.cpp` that's not already available by linking to the library?

coleharlow12, 2019-05-02:
That should have been obvious, I am still experimenting with and learning the LibPointMatcher Library. I assumed that ICP was meant to be usable outside of the library when clearly this was not the case. I apologize for the simple question, I am still new to programming. Thank You for your help

closed 2019-05-02 by coleharlow12

PhiBabin, 2019-05-02:
No need to apologize, have a nice day! 😉