norlab-ulaval/libpointmatcher #341: Compile on windows

lgilson, 2019-06-18:
Hi,

i had to tweek stuff in about 20 files (after CMAKE) in order to compile the lib on windows (Visual Studio 2017), but i have no idea why the linker is failing here:

public: virtual class Eigen::Matrix<float,-1,-1,0,-1,-1> __cdecl IdentityErrorMinimizer<float>::compute(struct PointMatcher<float>::ErrorMinimizer::ErrorElements const &)" (?compute@?$IdentityErrorMinimizer@M@@UEAA?AV?$Matrix@M$0?0$0?0$0A@$0?0$0?0@Eigen@@AEBUErrorElements@ErrorMinimizer@?$PointMatcher@M@@@Z)

Any pointers?

Thanks,

simonpierredeschenes, 2019-06-18:
@lgilson Hi, I just pushed a fix for Windows. However, this was just a one-line modification, I did not have to tweek 20 files. Also, I did not encounter linking problems with Eigen. Did you use the latest libpointmatcher version on the `master` branch and did you follow the compilation tutorial (https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md)? Also, what version of boost and eigen did you use?

lgilson, 2019-06-19:
Hi,

thanks for the quick reply. 

Yes, i followed that guide, but could not complete the "Build libpointmatcher" part because Boost is not found using the variable described (it works for libnabo...). That might be related to having 2 Visual Studios installed. I tried with a manual config, had to fix M_PI and various windows related details like the include you committed to end up with the linker error. 

I tried again the Cmake way by adding the keys BOOST_INCLUDEDIR and BOOST_LIBRARYDIR both pointing into the respective parts of boost. Cmake accepts this and the generated projects are valid. The Visual Studio 2017 build produced ~ 100k warnings (is that normal?), 0 errors. 

pmicp.exe runs correctly.

Thanks

closed 2019-06-19 by lgilson