norlab-ulaval/libpointmatcher #351: CMakeLists.txt and VC++

afabri, 2019-08-23:
* `add_definitions(-O3)`  triggers a warning for VC++ 
* `-Wall` creates too many warnings and should maybe not be the default.

PhiBabin, 2019-08-25:
We should change the optimization flag, if it's a Windows build. [This seem to be the modern way of setting compiler flags.](https://stackoverflow.com/questions/45955272/modern-way-to-set-compiler-flags-in-cross-platform-cmake-project)

As for `-Wall` can you be more precise about the kind of warning you want to be suppressed? `-Wall  -Wextra` is, as far as I know, the best practice for warning verbosity for C++. The problem we have with warning is most of them are not caused by libpointmatcher, but by Eigen. There are way of suppressing warning coming from Eigen.

afabri, 2019-08-26:
The warnings are not in Eigen, but triggered by how you use Eigen. When you write
`const int pointsCount(features.cols());` 
you get the warning
`warning C4244: 'initializing': conversion from 'Eigen::EigenBase<Derived>::Index' to 'int', possible loss of data`
So either you `static_cast` or use the right type. 

In my opinion there are too many warnings so Wall is not really helpful.

afabri, 2019-09-02:
You would have to write `add_definitions( /wd4244 /wd4267)`   but it then also silences warnings concerning your own code.