norlab-ulaval/libpointmatcher #352: Compilation with Visual C++ and Eigen master fails

afabri, 2019-08-23:
I ask as I get compilation errors (with the [master branch](https://github.com/eigenteam/eigen-git-mirror) of Eigen)

> 
C:\3rdPartyLibs\libpointmatcher\pointmatcher\DataPointsFilters\Gestalt.cpp(498): error C2676: binary '+=': 'Eigen::IndexedView<Derived,double,double>' does not define this operator or a conversion to a type acceptable to the predefined operator
        with
        [
            Derived=Eigen::Matrix<double,-1,-1,0,-1,-1>
        ]
C:\3rdPartyLibs\libpointmatcher\pointmatcher\DataPointsFilters\Gestalt.cpp(499): error C2675: unary '++': 'Eigen::IndexedView<Derived,double,double>' does not define this operator or a conversion to a type acceptable to the predefined operator
        with
        [
            Derived=Eigen::Matrix<double,-1,-1,0,-1,-1>
        ]
C:\3rdPartyLibs\libpointmatcher\pointmatcher\DataPointsFilters\Gestalt.cpp(514): error C2677: binary '-': no global operator found which takes type 'Eigen::IndexedView<Derived,double,double>' (or there is no acceptable conversion)
        with
        [
            Derived=Eigen::Matrix<double,-1,-1,0,-1,-1>
        ]

[Gestalt.cpp(498)](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/DataPointsFilters/Gestalt.cpp#L498)

ggael, 2019-08-23:
The problem is the use of a floating-point type to store and pass indices. Line 483, the variable `indices` should be declared as an `Int64Matrix` (or an `Eigen::MatrixXi`).

tkircher, 2020-07-02:
Thanks, this solved the same issue for me.

pomerlef, 2020-07-02:
It as been merged, so I'll close this issue. Thanks for reporting and solving!

closed 2020-07-02 by pomerlef