norlab-ulaval/libpointmatcher #353: Segment fault when convert ros msg to point matcher

LiShuaixin, 2019-08-23:
Hi, I tried to testing [segmap](https://github.com/ethz-asl/segmap) whose dependencies including libpointmatcher, but the system was killed by a segment fault caused at `PointMatcher_ros::rosMsgToPointMatcherCloud<float>(cloud_msg_in);`

I debugged the code and found the exact position of the fault is at line 159 in `pointmatcher_ros/point_cloud.cpp`: `case PF::FLOAT32: { view(dim, ptId) = T(*reinterpret_cast<const float*>(fPtr));`

When it try reassigning vals to `view`, which is a `Eigen::Block<Matrix>`, the segment fault occurred.

I can successfully `cout` the vals of `*reinterpret_cast<const float*>(fPtr)` which means the accessing of the array is valid. Moreover, at line 77 in the same .cpp file, when using  `cloud.getFeatureViewByName(it->name).setConstant(1);` to assign vals to the matrix, it is valid. But if `cout` the matrix after assignment, it is invalid. It's so weird!

As far as I know, if the declaration of a dynamic matrix with known size or other fixed-size matrix, the assignment and accessing of the matrix should be legal. Is this the problem of the version of Eigen?

PhiBabin, 2019-08-25:
The problem might be related to the version of eigen. Also, segmap is using an older version of `libpointmatcher` and `libpointmatcher_ros` ([https://github.com/ethz-asl/ethzasl_icp_mapping/commit/a7d8ea9](a7d8ea9)). I found the commit hash in [this file](https://github.com/ethz-asl/segmap/blob/master/dependencies.rosinstall). Maybe, updating this file, will update the dependencies?
[This is the list of changes](https://github.com/ethz-asl/ethzasl_icp_mapping/compare/a7d8ea9...master#diff-a6d1027369be6deca026699bd09e67e7) made to libpointmatcher_ros, since commit a7d8ea9.

LiShuaixin, 2019-08-26:
Hi @PhiBabin , thanks for your reply. Have you run segmap? which verision of eigen should I use to make the whole thing right?  I'll also try to fix the issue using the latest version of eigen

PhiBabin, 2019-08-26:
I have not run segmap. You simply have to make sure that every package is using the same Eigen version. So it's more about finding a different version of Eigen on your system, than trying to compile with the latest version of Eigen.

LiShuaixin, 2019-08-26:
Hi, thanks for your suggestion. I use Eigen 3.2 and check every packages in the project to ensure the consistency of Eigen version and the issue was solved.

closed 2019-08-26 by LiShuaixin