norlab-ulaval/libpointmatcher #357: How to get rotation angles from transformation matrix ?

rpadmanabhan, 2019-09-10:
Is there a function in the library which can be used to get the rotation angles ? 
I saw a reference here : https://github.com/ethz-asl/libpointmatcher/blob/bb59d2c70ace1d2201c02fa329044fc8fb2aab34/pointmatcher/ErrorMinimizers/PointToPlane.cpp#L208
Would using those formulae give the accurate answer ?

Does the rotation order follow : [w.r.t X in the X-Y plane ] -> [w.r.t Z in the Z-Y plane] -> [w.r.t. Z in the Z-X plane] ?

I am a novice in computer vision so sorry for the basic question.

Thanks

pomerlef, 2019-09-10:
There is not specific function in the library for that. You can code what is in comments in your own program or better, use the C++ Eigen functions for rotations: 

https://eigen.tuxfamily.org/dox/group__TutorialGeometry.html

Rotations are complicated and it would be too long to do a tutorial on them here. But, in general you shouldn't need to extract Euler angles from a rotation matrix. It will cause you more problems than help you on the long run.

rpadmanabhan, 2019-09-10:
Thanks, yes I understand that the math is complicated.
In my case the rotations should be small (max +- 0.1 radians), I just wanted to see what the angles are to get a sense of the magnitude of the tilts. 
I found another place where there is a reference to the angles, though the math is slightly different : https://github.com/ethz-asl/libpointmatcher/blob/b812b0958d3b65bad7c15716fe843fe752ad8d60/pointmatcher/TransformationChecker.cpp
Do you know which one would be more accurate to use for small angles ? Thank You

closed 2019-09-24 by rpadmanabhan