norlab-ulaval/libpointmatcher #358: ubuntu14.04 compile failed

debuleilei, 2019-09-19:
Hi I met a problem with compiling this source code in ubuntu 14.04 ,but success in ubuntu 16.04 .can you help me? thanks

[ 28%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/DataPointsFilters/SurfaceNormal.cpp.o
In file included from /usr/include/eigen3/Eigen/src/Core/MatrixBase.h:139:0,
                 from /usr/include/eigen3/Eigen/Core:279,
                 from /usr/include/eigen3/Eigen/StdVector:14,
                 from /home/caros/lib/libpointmatcher-master/pointmatcher/PointMatcher.h:47,
                 from /home/caros/lib/libpointmatcher-master/pointmatcher/DataPointsFilters/SurfaceNormal.h:37,
                 from /home/caros/lib/libpointmatcher-master/pointmatcher/DataPointsFilters/SurfaceNormal.cpp:35:
/usr/include/eigen3/Eigen/src/plugins/MatrixCwiseBinaryOps.h: In instantiation of ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<typename Eigen::internal::traits<T>::Scalar>, const Derived, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<typename Eigen::internal::traits<T>::Scalar>, Derived> > Eigen::MatrixBase<Derived>::cwiseMax(const Scalar&) const [with Derived = Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>; typename Eigen::internal::traits<T>::Scalar = float; Eigen::MatrixBase<Derived>::Scalar = float]’:
/home/caros/lib/libpointmatcher-master/pointmatcher/DataPointsFilters/SurfaceNormal.cpp:228:20:   required from ‘void SurfaceNormalDataPointsFilter<T>::inPlaceFilter(SurfaceNormalDataPointsFilter<T>::DataPoints&) [with T = float; SurfaceNormalDataPointsFilter<T>::DataPoints = PointMatcher<float>::DataPoints]’
/home/caros/lib/libpointmatcher-master/pointmatcher/DataPointsFilters/SurfaceNormal.cpp:292:17:   required from here
/usr/include/eigen3/Eigen/src/plugins/MatrixCwiseBinaryOps.h:110:72: error: could not convert ‘Eigen::MatrixBase<Derived>::cwiseMax(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> >; Derived = Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>; typename Eigen::internal::traits<T>::Scalar = float]((*(const Eigen::MatrixBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> > >*)(& Eigen::DenseBase<Derived>::Constant(Eigen::DenseBase<Derived>::Index, Eigen::DenseBase<Derived>::Index, const Scalar&) [with Derived = Eigen::Matrix<float, -1, 1>; Eigen::DenseBase<Derived>::ConstantReturnType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> >; typename Eigen::internal::traits<T>::Scalar = float; Eigen::DenseBase<Derived>::Index = long int; Eigen::DenseBase<Derived>::Scalar = float](((const Eigen::EigenBase<Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> >*)this)->Eigen::EigenBase<Derived>::cols<Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> >(), (* & other)))))’ from ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> > >’ to ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > >’
   return cwiseMax(Derived::PlainObject::Constant(rows(), cols(), other));
                                                                        ^
/usr/include/eigen3/Eigen/src/plugins/MatrixCwiseBinaryOps.h: In instantiation of ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_min_op<typename Eigen::internal::traits<T>::Scalar>, const Derived, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<typename Eigen::internal::traits<T>::Scalar>, Derived> > Eigen::MatrixBase<Derived>::cwiseMin(const Scalar&) const [with Derived = Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > >; typename Eigen::internal::traits<T>::Scalar = float; Eigen::MatrixBase<Derived>::Scalar = float]’:
/home/caros/lib/libpointmatcher-master/pointmatcher/DataPointsFilters/SurfaceNormal.cpp:228:20:   required from ‘void SurfaceNormalDataPointsFilter<T>::inPlaceFilter(SurfaceNormalDataPointsFilter<T>::DataPoints&) [with T = float; SurfaceNormalDataPointsFilter<T>::DataPoints = PointMatcher<float>::DataPoints]’
/home/caros/lib/libpointmatcher-master/pointmatcher/DataPointsFilters/SurfaceNormal.cpp:292:17:   required from here
/usr/include/eigen3/Eigen/src/plugins/MatrixCwiseBinaryOps.h:86:72: error: could not convert ‘Eigen::MatrixBase<Derived>::cwiseMin(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> >; Derived = Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > >; typename Eigen::internal::traits<T>::Scalar = float]((*(const Eigen::MatrixBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> > >*)(& Eigen::DenseBase<Derived>::Constant(Eigen::DenseBase<Derived>::Index, Eigen::DenseBase<Derived>::Index, const Scalar&) [with Derived = Eigen::Matrix<float, -1, 1>; Eigen::DenseBase<Derived>::ConstantReturnType = Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> >; typename Eigen::internal::traits<T>::Scalar = float; Eigen::DenseBase<Derived>::Index = long int; Eigen::DenseBase<Derived>::Scalar = float](((const Eigen::EigenBase<Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > > >*)this)->Eigen::EigenBase<Derived>::cols<Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > > >(), (* & other)))))’ from ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_min_op<float>, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > >, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Matrix<float, -1, 1> > >’ to ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_min_op<float>, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > >, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::CwiseBinaryOp<Eigen::internal::scalar_max_op<float>, const Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true>, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<float>, Eigen::Block<Eigen::Block<Eigen::Matrix<float, -1, -1>, -1, -1, false>, -1, 1, true> > > > >’
   return cwiseMin(Derived::PlainObject::Constant(rows(), cols(), other));
                                                                        ^
[... cut ...] >, const Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Block<Eigen::Block<Eigen::Matrix<double, -1, -1>, -1, -1, false>, -1, 1, true> > > > >’
   return cwiseMin(Derived::PlainObject::Constant(rows(), cols(), other));
                                                                        ^
make[2]: *** [CMakeFiles/pointmatcher.dir/pointmatcher/DataPointsFilters/SurfaceNormal.cpp.o] Error 1
make[1]: *** [CMakeFiles/pointmatcher.dir/all] Error 2
make: *** [all] Error 2

simonpierredeschenes, 2019-09-19:
This seems to be related to Eigen. Could you report the output you get when you run the script utest/listVersionsUbuntu.sh

debuleilei, 2019-09-21:
Copy-paste those information when reporting a bug in libpointmatcher:

Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 14.04.6 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 4.8.4-2ubuntu1~14.04.4) 4.8.4
git: 		| git version 1.9.1
cmake: 		| cmake version 2.8.12.2
boost: 		| Version: 1.54.0.1ubuntu1
eigen3: 	| Version: 3.2.0-8
dpkg-query: package 'doxygen' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
doxygen: 	|

simonpierredeschenes, 2019-09-21:
You could try to compile the latest release (commit eaece708ceda032501568563a91eeb3d71a5ca8d). The SurfaceNormalDataPointsFilter has been modified since then. This might be the cause.

debuleilei, 2019-09-23:
yeah, it's ok now ,thank you!

closed 2019-10-12 by PhiBabin