norlab-ulaval/libpointmatcher #359: Exception thrown when using PureTranslation

rpadmanabhan, 2019-10-01:
Hi,
I get this exception from TransformationsImpl.cpp when using the PureTranslation transformation :
```
if(this->checkParameters(parameters) == false)
    throw PointMatcherSupport::TransformationError("PureTranslation: Error, left part  not identity.");
```

The error seems to suggest the transformation matrix is not as expected as that for a PureTranslation. 
I am using the api like this : 
```
PM::ICP icp_;

icp_.transformations.push_back(PM::get().TransformationRegistrar.create("PureTranslation"));

icp_.matcher = PM::get().MatcherRegistrar.create("KDTreeMatcher", PM::Parameters{ {"knn", toParam(1)} });

icp_.outlierFilters.push_back(PM::get().OutlierFilterRegistrar.create("RobustOutlierFilter"));

// ErrorMinimizer - PointToPoint
icp_.errorMinimizer = PM::get().ErrorMinimizerRegistrar.create("PointToPointErrorMinimizer");

// Inspector - PerformanceInspector
icp_.inspector = PM::get().InspectorRegistrar.create("PerformanceInspector");

PM::TransformationParameters T = icp_(rx_points, r1_points); // get the transformation matrix
```

Any idea what might be going wrong ?

Thank You

PhiBabin, 2019-10-03:
The behavior of `icp_.transformations` is a bit weird, it's goal is to assure to consistency of the transformation matrices that are computed. It's goal is not to convert rigid transformation to a target transformation or to enforce restrictions on the minimization. `icp_.transformations` is used in two places: 1) for applying the initial estimate (here it's an identity) 2) for applying the transformation computed by the minimizer (here the `PointToPointErrorMinimizer`). Since this minimizer returns a rigid transformation, the `PureTranslation` will detect that the transformation matrix has a rotation part which is not identity.

If your goal is to limit the ICP minimization to only minimize translation, I suggest creating your own PointToPoint minimizer that only minimize translation. This should be easy to implement since the minimum translation will always be the average distance between the matched pairs (be aware that all distances in libpointmatcher are the squared distance). If you implement it, please do a pull request ;)

rpadmanabhan, 2019-10-04:
Thanks ! I understand the issue now. I will definitely send a PR if i implement a minimizer for only the translations.

closed 2019-10-04 by rpadmanabhan