norlab-ulaval/libpointmatcher #361: how to init reference points once, and then don't update???

chennuo0125, 2019-10-12:
when i use libpointmatcher to Robot Localization, matching processing cost too much time. Robot Localization match reading points to fixed and large map(refer points), so no need to init map for each matching process,  maybe which can decrease the cost time????

PhiBabin, 2019-10-12:
I am not sure I understand, by 'init reference points once', do you mean run the reference's datapoints filters only once during a run of ICP? Or do you mean that you don't need a matching step, since your 'Robot Localization' already know the correspondence between the reading and reference points?
Or do you mean that you don't need to initialize the KD-tree of the reference point at each run of ICP, but instead only do it when the map is updated?

If you question was the last one use the `ICPSequence` instead of `ICP`, it will only run the reference datapoint filters and will only initiate the KD-tree when you call `ICPSequence::setMap()`.

chennuo0125, 2019-10-12:
my question was the last one, and i has solve the problem follow your answer, thank you :) :) :)

closed 2019-10-12 by chennuo0125