norlab-ulaval/libpointmatcher #362: Error during compilation: No match for operator

mstampa, 2019-10-16:
Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 18.04.3 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
git: 		| git version 2.17.1
cmake: 		| cmake version 3.10.2 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: 		| Version: 1.65.1.0ubuntu1
eigen3: 	| Version: 3.3.4-4
doxygen: 	| Version: 1.8.13-10

Hello folks,

I'm trying to use libpointmatcher within a ROS workspace as a dependency to [SegMapper](https://github.com/ethz-asl/segmap/). Next to a couple of harmless warnings regarding std::auto_ptr being deprecated, I get errors of the following form during compilation:

`~/work/segmap_ws/src/libpointmatcher/pointmatcher/DataPointsFilters/Gestalt.cpp:498:52: error: no match for ‘operator+=’ (operand types are ‘Eigen::internal::enable_if<true, Eigen::IndexedView<Eigen::Matrix<float, -1, -1>, float, float> >::type {aka Eigen::IndexedView<Eigen::Matrix<float, -1, -1>, float, float>}’ and ‘Eigen::DenseCoeffsBase<Eigen::Matrix<float, -1, 1>, 1>::Scalar {aka float}’)
         gestaltMeans(indices(0,it), indices(1,it)) += heights(it);
`

Similar errors occur for operator++ and operator-  in other lines of the same file. 

Issue #329 describes the same problem, but was never fully answered. As proposed by @simonpierredeschenes there I reinstalled Eigen (`sudo apt install --reinstall libeigen3-dev; sudo dpkg-reconfigure libeigen3-dev`), unfortunately to no avail. 
Other things I've tried:
- Check out the master branch (SegMap lists an older version as its dependency)
- Compile with C++ standards 11,14 and 17: `catkin build -DCMAKE_CXX_STANDARD=14 libnabo libpointmatcher`

Hopefully I didn't miss anything too obvious. Any help for fixing this problem will be highly appreciated!

Thanks in advance.

paulacarbo07, 2020-01-28:
Hi! I am having this issue as well, trying to compile it as a dependency of SegMap. Did you manage to solve it in the end? I would very much appreciate any guidance.

pomerlef, 2020-01-28:
@simonpierredeschenes can you have a look?

mstampa, 2020-01-29:
I have a feeling that the dependency [eigen_catkin](https://github.com/ethz-asl/eigen_catkin) might be the root problem. I replaced it with "eigen" in all package.xml files, and had to adapt some CMakeLists.txt and include-directives here and there. I did the same with pcl_catkin, doxygen_catkin etc, essentially trying to use versions shipped with Ubuntu and ROS instead of compiling these packages myself whenever possible.

However, I wasn't smart enough to document my changes. In the end, I was able to compile SegMap but it still crashes immediately after startup. But this might be an issue to open up on their page rather than here.

If I have some free time I will try to recreate what I did, but of course I'd prefer if the developers had a look.

simonpierredeschenes, 2020-01-29:
I will try to have a look soon. However, it might take some time before I figure out the issue myself. I will keep you updated if I find something.

mstampa, 2020-02-07:
@simonpierredeschenes I was able to recreate what I did for libpointmatcher. In short: 
- Use find_package for Eigen3 in CMakeLists.txt (it was already included as a TODO)
- Update all eigen include paths, e.g. from `#include <Eigen/Core>` to `#include <eigen3/Eigen/Core>` I had to do this once in libnabo/nabo/nabo.h too.

On my system it compiles without errors. Please see the patch file attached. Hope it helps.
[issue362.txt](https://github.com/ethz-asl/libpointmatcher/files/4171171/issue362.txt)

fangcywangermu, 2020-03-06:
> @simonpierredeschenes我能够重新创建我为libpointmatcher做的事情。简而言之：
> 
> * 在CMakeLists.txt中对Eigen3使用find_package（它已经作为TODO包含在内）
> * 更新所有特征包含路径，例如从`#include <Eigen/Core>`到`#include <eigen3/Eigen/Core>`。libnabo / nabo / nabo.h中也有此必要。
> 
> 请参阅附件中的补丁文件。在我的系统上，它现在可以正确编译，希望对您有所帮助！
> [issue362.txt](https://github.com/ethz-asl/libpointmatcher/files/4171171/issue362.txt)

我按照您说的更改之后还是出现不匹配的问题，按照txt里面说的都更改了啊

pomerlef, 2020-03-06:
@fangcywangermu in English would be better...

fangcywangermu, 2020-03-06:
> @simonpierredeschenes I was able to recreate what I did for libpointmatcher. In short:
> 
> * Use find_package for Eigen3 in CMakeLists.txt (it was already included as a TODO)
> * Update all eigen include paths, e.g. from `#include <Eigen/Core>` to `#include <eigen3/Eigen/Core>`. This is also necessary in libnabo/nabo/nabo.h.
> 
> Please see the patch file attached. On my system it compiles without errors now, hope it helps!
> [issue362.txt](https://github.com/ethz-asl/libpointmatcher/files/4171171/issue362.txt)

After I changed according to what you said, there was still a problem of mismatch. I changed it according to what txt said.

fangcywangermu, 2020-03-06:
> @fangcywangermu in English would be better...

OK

bunalti, 2023-01-28:
I am getting a quite similar (double not flaot) error on a ROS2 Grid Mapper code. I checked the attached file issue362.txt. Did the exact changes but still no use. I use eigen3.4.0 and ROS2 humble.

bunalti, 2023-01-31:
I have changed my libeginen3-dev package version=3.4.0  to version=3.3.4 and version=3.3.7 both seems to compile without any issues.

closed 2023-01-31 by pomerlef