norlab-ulaval/libpointmatcher #363: how to calculate mean dist when use ICPSequence ???

chennuo0125, 2019-10-22:
when i use icp object to calculate mean dist as follows:  
```  
double BotLocalization::getMeanDistPointMatcher(const DP& ref_points, const DP& transformed_src_points)
{
    //for retrieving paired point mean distance without outliers. We reuse the same module used for the icp object.
    PM::Matches matches;
    icp_.matcher->init(ref_points);
    matches = icp_.matcher->findClosests(transformed_src_points);// extract closest points	
    const PM::OutlierWeights outlierWeights = icp_.outlierFilters.compute(transformed_src_points, ref_points, matches);// weight paired points
    // generate tuples of matched points and remove pairs with zero weight
    const PM::ErrorMinimizer::ErrorElements matchedPoints( transformed_src_points, ref_points, outlierWeights, matches);
    // extract relevant information for convenience
    const int dim = matchedPoints.reading.getEuclideanDim();
    const int nbMatchedPoints = matchedPoints.reading.getNbPoints(); 
    const PM::Matrix matchedRead = matchedPoints.reading.features.topRows(dim);
    const PM::Matrix matchedRef = matchedPoints.reference.features.topRows(dim);
    // compute mean distance
    const PM::Matrix dist = (matchedRead - matchedRef).colwise().norm(); // replace that by squaredNorm() to save computation time
    const float meanDist = dist.sum()/nbMatchedPoints;
    return meanDist;
}    
```
the mean dist result is ok, but when i use icpsequence to calculate mean dist as follows:  
```
double BotLocalization::getMeanDistPointMatcher(const DP& ref_points, const DP& transformed_src_points)
{
    //for retrieving paired point mean distance without outliers. We reuse the same module used for the icp object.
    PM::Matches matches;
    matches = icpSequence_.matcher->findClosests(transformed_src_points);// extract closest points	
    const PM::OutlierWeights outlierWeights = icpSequence_.outlierFilters.compute(transformed_src_points, ref_points, matches);// weight paired points
    // generate tuples of matched points and remove pairs with zero weight
    const PM::ErrorMinimizer::ErrorElements matchedPoints( transformed_src_points, ref_points, outlierWeights, matches);
    // extract relevant information for convenience
    const int dim = matchedPoints.reading.getEuclideanDim();
    const int nbMatchedPoints = matchedPoints.reading.getNbPoints(); 
    const PM::Matrix matchedRead = matchedPoints.reading.features.topRows(dim);
    const PM::Matrix matchedRef = matchedPoints.reference.features.topRows(dim);
    // compute mean distance
    const PM::Matrix dist = (matchedRead - matchedRef).colwise().norm(); // replace that by squaredNorm() to save computation time
    const float meanDist = dist.sum()/nbMatchedPoints;
    return meanDist;
}
```
the result isnt ok, i couldnt konw why ? please give me some advise, thank you !

chennuo0125, 2019-10-22:
icpSequence object has initialed use interface :   
```
icpSequence_.setMap(ref_points_)
```

PhiBabin, 2019-10-26:
I don't know if it solves your problem, but you can simplify your code by doing this:
```
    PM::Matches matches;
    matches = icpSequence_.matcher->findClosests(transformed_src_points);// extract closest points	
    const PM::OutlierWeights outlierWeights = icpSequence_.outlierFilters.compute(transformed_src_points, ref_points, matches);// weight paired points
    const PM::ErrorMinimizer::ErrorElements matchedPoints( transformed_src_points, ref_points, outlierWeights, matches);
    const int nbMatchedPoints = matchedPoints.reading.getNbPoints(); 
    float meanDist = matchedPoints.matches.dists.sum() / nbMatchedPoints;
```

If that does not change anything, try reloading the configuration of ICP. Maybe, you have stumble upon a case where with ICPSequence, the state of the matcher is not correctly reset when you directly call it.

PhiBabin, 2019-10-26:
I think I found the problem. Inside ICP for optimization reasons, the origin of the reference and the reading are the center of mass of the reference (the average value of the point). The map contains inside the KdTree of ICPSequence has for origin the center of mass of the map. When you register a reading against the map, the functions inside ICP change the origin of the reading before calling the KdTree.
So your second example you try to match your reading point cloud (which is in the map frame) against the KdTree of the map (which is in a center of mass frame). One way solving your problem is to move your reading to the center of mass frame, like [here](https://github.com/ethz-asl/libpointmatcher/blob/b812b0958d3b65bad7c15716fe843fe752ad8d60/pointmatcher/ICP.cpp#L329)

chennuo0125, 2019-10-26:
i see that T_refIn_refMean is protected, but i dont want to change the libpointmatcher library, how to get T_refIn_refMean and T_refIn_dataIn by other interface or methods ???

PhiBabin, 2019-10-26:
You could inherit from `ICPSequence` and add a `getAverageDist()` method which uses those attributes.

chennuo0125, 2019-10-27:
thank you, i have get the expected results by follow your advise. :)  :)  :)

PhiBabin, 2019-10-27:
Happy to help!

closed 2019-10-27 by PhiBabin