norlab-ulaval/libpointmatcher #364: Can't get PointToPlaneErrorMinimizer to work

tfwittwer, 2019-11-05:
Hi,

I'm trying to do point-to-plane ICP with the pmicp program. My dataset is a cross section extracted from mobile laser scan of a piece of track. I've tried several different data filters, but I never get any output. The furthest I've gotten is the 

PointMatcher::PointToPlaneErrorMinimizer - minimization will be in 2D.

message. Point to Point works fine. 
I'm assuming it is because not enough planar features can be found in the dataset, but I don't see any way of tuning what is to be considered a plane. What can I do differently to get point-to-plane to work?

Regards,

Tobias

pomerlef, 2019-11-05:
Can you send your configuration file (`yaml`)?

tfwittwer, 2019-11-06:
This is one of the configurations I tried:

readingDataPointsFilters:
    IdentityDataPointsFilter

referenceDataPointsFilters:
    IdentityDataPointsFilter

matcher:
  KDTreeMatcher:
    knn: 1

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.95

errorMinimizer:
  PointToPlaneErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 100
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.001
      minDiffTransErr: 0.001
      smoothLength: 4 

inspector:
  NullInspector

logger:
  FileLogger

PhiBabin, 2019-11-06:
You are missing the `SamplingSurfaceNormalDataPointsFilter` filter to generate the normal on the reference point cloud. [Check out this example yaml configuration.](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/data/default.yaml)

tfwittwer, 2019-11-06:
I've tried using the example yaml, but I'm not getting any results. This is the program output:

* KDTreeMatcher: initialized with knn=1, epsilon=0, searchType=1 and maxDist=inf
PointMatcher::PointToPlaneErrorMinimizer - minimization will be in 2D.
Applying 1 DataPoints filters - 211005 points in
* SamplingSurfaceNormalDataPointsFilter - 105276 points out (-50.1073%)
Applied 1 filters - 105276 points out (-50.1073%)
PointMatcher::icp - reference pre-processing took 0.339 [s]
Applying 1 DataPoints filters - 138635 points in
* RandomSamplingDataPointsFilter - 69243 points out (-50.0537%)
Applied 1 filters - 69243 points out (-50.0537%)
PointMatcher::icp - reading pre-processing took 0.009 [s]
PointMatcher::icp - 5 iterations took 0.124 [s]
match ratio: 0.750011

Configuration:
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.5

referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 10

matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 0 

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.75

errorMinimizer:
  PointToPlaneErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 40
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.001
      minDiffTransErr: 0.01
      smoothLength: 4   

inspector:
  NullInspector

logger:
  FileLogger

pomerlef, 2019-11-06:
To receive the message `PointMatcher::PointToPlaneErrorMinimizer - minimization will be in 2D.` the parameter `force2D` needs to be set to `true` somewhere.

In your yaml file, try:
```
errorMinimizer:
   PointToPlaneErrorMinimizer:
      force2D: 0
```

When you say, 

> not getting any results

you mean that no transformation is printed or that you're not getting the transformation you were expecting?

tfwittwer, 2019-11-06:
I've tried the force2D: 0 parameter, but still get the message about minimization being in 2D.

I get no result files at all. The program seems to be quitting prematurely, as I have added the computation and output of the fitting error and don't see that when using point-to-plane.

pomerlef, 2019-11-06:
Alright, that's not normal.

@tfwittwer : please share with us your two point clouds and config.yaml

@simonpierredeschenes can you have a look to the data?

tfwittwer, 2019-11-07:
This is the download link: https://we.tl/t-6wR2kF0bR8

Thanks for your help!

tfwittwer, 2019-11-12:
I see that the data has not been downloaded yet. Should I share it in some other way?

simonpierredeschenes, 2019-11-12:
I will have a look today, sorry for the delay!

simonpierredeschenes, 2019-11-12:
When I run pmicp with the following command:
```
./pmicp --config icp_cfg.yaml 181215_015439_S1.csv 181215_022701_S1.csv
```

the three output files (test_ref.vtk, test_data_in.vtk and test_data_out.vtk) are generated. I use the latest version (eb9457d) and the config I use is:

```
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.5

referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 10

matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 0

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.75

errorMinimizer:
  PointToPlaneErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 40
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.001
      minDiffTransErr: 0.01
      smoothLength: 4

inspector:
  NullInspector

logger:
  FileLogger
```

The "minimization will be in 2D" message is always printed, whatever the config is, but it does not mean that minimization is always done in 2D. I will open an issue for this logging problem.

What is outputed if you run the `libpointmatcher/utest/listVersionsUbuntu.sh` script?

tfwittwer, 2019-11-12:
I haven't tried libpointmatcher/utest/listVersionsUbuntu.sh yet (I'm on Windows, by the way - not sure if that may be the cause of the problems), but running your command and yaml file does not produce the vtk files. The console output is

* KDTreeMatcher: initialized with knn=1, epsilon=0, searchType=1 and maxDist=inf
PointMatcher::PointToPlaneErrorMinimizer - minimization will be in 2D.
Applying 1 DataPoints filters - 211005 points in
* SamplingSurfaceNormalDataPointsFilter - 105276 points out (-50.1073%)
Applied 1 filters - 105276 points out (-50.1073%)
PointMatcher::icp - reference pre-processing took 0.337 [s]
Applying 1 DataPoints filters - 138635 points in
* RandomSamplingDataPointsFilter - 69243 points out (-50.0537%)
Applied 1 filters - 69243 points out (-50.0537%)
PointMatcher::icp - reading pre-processing took 0.01 [s]
PointMatcher::icp - 5 iterations took 0.123 [s]

tfwittwer, 2019-11-13:
Ok, stepping through the program with the debugger has given me more insight. As I wrote above, I've modified the pmicp program to give me the residual error, as described here: https://github.com/ethz-asl/libpointmatcher/issues/193#issue-203885636

This line causes an exception:
float error = icp.errorMinimizer->getResidualError(data_out, ref, outlierWeights, matches);

Unhandled exception at 0x00007FFA863D9129 in pmicp.exe: Microsoft C++ exception: PointMatcher<float>::DataPoints::InvalidField at memory location 0x0000000BD9D8E758.

This does not occur with the PointToPlaneErrorMinimizer

PhiBabin, 2019-11-14:
Can you get the exception message of the InvalidField exception? Normally, when an exception is uncaught it is suppose to print the error message in the console.
Before trying to get the residual error working, it would be preferable to fix the problem you have with ICP.
To debug the issue I suggest changing the inspector in the yaml configuration to:
```
inspector:
 VTKFileInspector:
     baseFileName: myoutput
     dumpPerfOnExit: 0
     dumpStats: 0
     dumpIterationInfo: 1 
     dumpDataLinks: 1 
     dumpReading: 1 
     dumpReference: 1 
```
Based on your previous log output, it seems that ICP, however the file are either not written on disk or written in the wrong folder. Adding this inspector, will make ICP write a vtk for each ICP iteration. Thus since your ICP execute 5 iterations, it should write something.

tfwittwer, 2019-11-14:
The ICP is working now. The program crashed before output was generated due to the error computation not working.

There is no error message output to the console. The exception message is:
Unhandled exception at 0x00007FFA863D9129 in pmicp.exe: Microsoft C++ exception: PointMatcher::DataPoints::InvalidField at memory location 0x0000000BD9D8E758.

I believe this is the same issue as reported in issue #193 (points have no normals).

PhiBabin, 2019-11-14:
Have tried doing the same solution as #193? The problem with his code was that he needed to apply the data point filter. He assumed that the datapoints where pass by reference, not by value.
The more efficient of computing the residual is to use `ICPSequence`. Check this #363 to get an idea of how to do it:

tfwittwer, 2019-11-15:
I don't see in #193 how exactly the data point filter is applied. I've taken a look at #363, but it is not clear to me which changes I would have to make to pmicp to make it work. It's pretty much a black box for me.

PhiBabin, 2019-11-15:
The error message said that your reference pointcloud does not have a normal field. In libpointmatcher normal are computed using a datapoint filter. The datapoint filters are set by loading a yaml configuration file. In #193, the datapoint `dpref` passed to this function does not have normals:
```c++
void getResidualError(const DP &dpref, const DP &dppointcloud_out) {
  icp_.matcher->init(dpref);
  PM::Matches matches = icp_.matcher->findClosests(dppointcloud_out);
  PM::OutlierWeights outlierWeights = icp_.outlierFilters.compute(dppointcloud_out, dpref, matches);
  float error = icp_.errorMinimizer->getResidualError(dppointcloud_out, dpref, outlierWeights, matches);
  std::cout << "Final residual error: " << error << std::endl;
}
```
You need to execute the reference's datapoint filters by doing the following:
```c++
icp_.referenceDataPointsFilters.apply(dpref);
getResidualError(dpref, dppointcloud_out);
```
`dpref`should now contain normals.

tfwittwer, 2019-11-19:
Got it to work, thank you very much!

closed 2019-11-19 by tfwittwer