norlab-ulaval/libpointmatcher #366: How to use libpointmatcher from Eigen Matrices (and not csv/vtk)?

jbmouret, 2019-11-22:
Hello

I might have missed something but all the tutorials only cover loading data from files. I want to use libpointmatcher directly with Eigen Nx3 matrices (each line is a 3D point).

By trial-and-error / reading the code, I tried this:
```C++ 

Eigen::MatrixXd merge_points(const Eigen::MatrixXd& pcl1, const Eigen::MatrixXd& pcl2)
{
    typedef PointMatcher<double> PM;
    typedef PM::DataPoints DP;

    DP ref;
    ref.features = Eigen::MatrixXd(pcl1.cols() + 1, pcl1.rows());
    ref.features << pcl1.transpose(), Eigen::VectorXd::Ones(pcl1.rows());
    
    DP data;
    data.features = Eigen::MatrixXd(pcl2.cols() + 1, pcl2.rows());
    data.features << pcl2.transpose(), Eigen::VectorXd::Ones(pcl2.rows());

    PM::ICP icp;
    icp.setDefault();
    PM::TransformationParameters T = icp(data, ref);

    DP data_out(data);
    icp.transformations.apply(data_out, T);

    Eigen::MatrixXd res(pcl1.rows() + pcl2.rows(), pcl1.cols()); 
    res << pcl1, data_out.features.topRows(3).transpose();

    std::cout << "Final transformation:" << std::endl << T << endl;

    return res;
}
```
... it runs but the transformation is wrong (and not the same as when I dump the Eigen matrices to csv files and load them to icp_simple).

=> What is the  recommended way to do this?

Thanks!

regards,
-- Jean-Baptiste

jbmouret, 2019-11-22:
Update: I get the right transformation if I transpose the Ones vectors:

```c++
 DP ref;
 ref.features = Eigen::MatrixXd(pcl1.cols() + 1, pcl1.rows());
  ref.features << pcl1.transpose(), Eigen::VectorXd::Ones(pcl1.rows()).transpose();
  DP data;
  data.features = Eigen::MatrixXd(pcl2.cols() + 1, pcl2.rows());
  data.features << pcl2.transpose(), Eigen::VectorXd::Ones(pcl2.rows()).transpose();
```

... but this still feels 'hacky' to me (espcially writing in `features` directly).

simonpierredeschenes, 2019-11-22:
In libpointmatcher, points are represented as column vectors, so you need to add a transpose somewhere when you create the cloud. Here is a simpler way to create a cloud (also less 'hacky'):
```
DP ref;
ref.addFeature("x", pcl1.col(0).transpose());
ref.addFeature("y", pcl1.col(1).transpose());
ref.addFeature("z", pcl1.col(2).transpose());
```
Hope it helps!

jbmouret, 2019-11-22:
thanks! this looks better.

closed 2019-11-22 by jbmouret