norlab-ulaval/libpointmatcher #368: Using libpointmatcher for 2D lidars extrinsic calibration

artemiialessandrini, 2019-12-21:


simonpierredeschenes, 2019-12-21:
I am not sure to understand what the problem is, do you get an output translation of 0 ([0, 0, 0]^T)? If you add a VTKFileInspector inspector to your yaml configuration file, libpointmatcher will save the reading point cloud at each iteration. Then, you will be able to see what is happening when the clouds are matched. The inspector configuration looks like this : 
```
VTKFileInspector:
    baseFileName: SOME_OUTPUT_PATH
    dumpReading: 1
    dumpReference: 1
    writeBinary: 1
```
If you cannot find the issue with the inspector, it would help if you could provide an example cloud to help us reproduce the issue on our side.

artemiialessandrini, 2019-12-23:
Thank you, @simonpierredeschenes , for quick response, I'll toggle an inspector on to visualize.

I expect matcher_service from ethzasl_icp_mapper to provide a `geometry_msgs/Transform` between two coordinate frames of PointClouds.
Here's my [bag file](https://yadi.sk/d/Wzu8Bit1Ud914Q), to reproduce the issue, please run:

`rosbag play non_linear.bag`
`rosrun ethzasl_icp_mapper matcher_service  _config:=<path>/<to>/icp.yaml`
`rosrun ethzasl_icp_mapper wrapper`

Wrapper client could be found [here](https://gist.github.com/artemiialessandrini/e06a6c0d7f7d0bea93169f3a9fe197de),
[That's](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/reintegrate/master_into_indigo_devel/ethzasl_icp_mapper/src/matcher_service.cpp) where the matcher_service is,
.yaml file configuration is attached on the top.