norlab-ulaval/libpointmatcher #369: Plans to implement new registration methods

YoshuaNava, 2020-01-13:
Hi,
I'm a frequent user of libpointmatcher, its filter chains and specially the point-to-plane error minimizer for ICP.

In recent times, new registration methods have been developed, which show really strong performance in benchmarks (e.g. KITTI). Most of them are variants of ICP depending on different descriptors, point representations, and introduce extra steps to ensure higher robustness. Some examples are: 
* ([GICP](http://www.roboticsproceedings.org/rss05/p21.pdf)
* [LOAM](https://www.ri.cmu.edu/publications/laser-visual-inertial-odometry-and-mapping-with-high-robustness-and-low-drift/) and [LeGo-LOAM](https://ieeexplore.ieee.org/document/8594299)
* [Auto-tuned ICP](https://ori.ox.ac.uk/where-did-my-robot-go-being-robust-and-cautious-during-laser-based-localization/)).

I was wondering if there are any plans to integrate any of the above methods into libpointmatcher in the near future, and if the main authors of the library have experience with them. I think libpointmatcher, as it is right now, must be the most flexible and feature rich implementation of ICP and point cloud filters, and could benefit from some of the above methods.

Thank you in advance.
https://frc.ri.cmu.edu/~zhangji/publications/ICRA_2016.pdf

pomerlef, 2020-01-13:
Yes, we are tracking those solutions (and others). The short terms goals are:
- NDT (point-to-Gaussian and Gaussian-to-Gaussian)
- GICP (plane-to-plane)
We are slowly finishing the math to have them efficiently optimized.

Auto-tuned ICP was relying on libpointmatcher for there publication, but I'm not aware that they send us a PR to integrate it.

LOAM (and alike) are tightly coupled with IMU (or vision). The short term step is to focus on the point-to-edge they were using.

YoshuaNava, 2020-01-13:
@pomerlef Thank you for your quick reply.

I would be interested in the developments, I'll stay in sync to help review. 

Regarding LOAM, I gather they use a different type of descriptor instead of surface normals, which is inspired on surface smoothness. Based on that they extract corner and edge points, and use them for matching. The data association is performed with k-NN, and they employ bi-square weights to account for outliers. What is your opinion about these changes?

Regarding LOAM's use of IMU data, it's interesting that they do it to accumulate point cloud data and release some burden from the frame-to-map localization module, but they don't employ the gravity vector to prevent the map from curving away. However, there are other methods (e,g, [LIO-mapping](https://github.com/hyye/lio-mapping)) which take advantage of widely available pre-integrated IMU graph factors to get a gravity-aligned localization estimate.

(I'm also potentially interested in contributing when it comes to GICP/LOAM, but for this I need to ask my supervisor first)