norlab-ulaval/libpointmatcher #372: Poor registration results

tfwittwer, 2020-04-15:
Hi,

I'm using the pmicp program to compute differences between multiple drivelines acquired with a mobile laser scanner on railroad track. I cut cross sections at suitable locations, e.g. catenary poles. I've tried many different parameters, but still struggle to get good results in X and Y direction (Z/height is Ok). 

This is one of the parameter sets that I have tried. I have attached a top view image of the registration result, which shows the X/Y shift. I'd be very interest in how to improve the registration. Thanks for your help!

Regards,

Tobias

[image]
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.5

referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 10

matcher:
  KDTreeMatcher:
    knn: 2
    epsilon: 0 

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.95

errorMinimizer:
  PointToPlaneErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 40
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.001
      minDiffTransErr: 0.01
      smoothLength: 4   

inspector:
  NullInspector

logger:
  FileLogger

pomerlef, 2020-04-15:
Send us your point clouds, @vdsmax will have a look.

We typically don't do application oriented support as there are many users and we have a limited time.

tfwittwer, 2020-04-15:
I've attached one sample with four files at the same location. Thanks for the effort, I appreciate it very much!

[055-29.879.zip](https://github.com/ethz-asl/libpointmatcher/files/4482810/055-29.879.zip)

vdsmax, 2020-04-15:
I have tested your parameters with the csv files you give me.
This is one result i obtain by doing a registration between 200402_075853_S1.csv and 200402_112838_S1.csv 
Is-it the desired result ?
What command did you use on your terminal with pmicp ?

[Registration_075853_112838.pdf](https://github.com/ethz-asl/libpointmatcher/files/4483573/Registration_075853_112838.pdf)

tfwittwer, 2020-04-16:
It looks good, but how does it fare with the other files? I get the biggest mismatch between 200402_075853_S1.csv and 200402_102843_S1.csv (the green and blue point cloud in my earlier post). My command line arguments are 
pmicp -v --config icp_cfg.yaml --isTransfoSaved true --output result 200402_075853_S1.csv 200402_102843_S1.csv

vdsmax, 2020-04-16:
I have done the registration between all the files and i obtained the same good result with your command line.
Can you show me the message in the terminal when you execute the command line ?

tfwittwer, 2020-04-16:
There's no message in the command line. What translation vector do you get? I get 0.116591/-0.0611682/-0.0611682 for the above parameter set and point clouds, which results in the shift shown above.
My version of libpointmatcher is a couple of months old, have there been changes that may improve the results? I'm on Windows, by the way.

vdsmax, 2020-04-16:
I have the same messages. What is your transform result matrix ?

tfwittwer, 2020-04-16:
0
0
0
    0.999997 -0.000118906  -0.00258197    0.0581731
 0.000122226     0.999999    0.0012908  -0.00572896
  0.00258182  -0.00129114     0.999996   -0.0212364
           0            0            0            1

vdsmax, 2020-04-16:
I have this matrix transform between 200402_075853_S1.csv and 200402_102843_S1.csv:
0.999996 -0.000332147  -0.00280515    0.0631777
 0.000335855     0.999999   0.00131125  -0.00533485
  0.00280472  -0.00131219     0.999995   -0.0207462
           0            0            0            1
It's very close to the one you have. What display software did you use to see the point cloud ?

tfwittwer, 2020-04-16:
I use a program that I've written myself based on OpenTK, and CloudCompare.

With this vector, the registration won't be very good. 200402_102843_S1 needs to be shifted to the left and up for good registration with 200402_075853_S1, not to the right and down as computed by both of us.

[image]

vdsmax, 2020-04-16:
We use Paraview to see your PointClouds. With this software, i can see the shift between your two scans, 200402_075853_S1 and 200402_102843_S1, before the registration as shown in the figure below:

[shift_X_and_Z_before_registration.pdf](https://github.com/ethz-asl/libpointmatcher/files/4487805/shift_X_and_Z_before_registration.pdf)

I computed the transform given by libpointmatcher in Paraview and the result is shown in the figure below:

[after_registration.pdf](https://github.com/ethz-asl/libpointmatcher/files/4487811/after_registration.pdf)

I don't have your shift. Verify that you have the same result than me in Paraview first.

tfwittwer, 2020-04-16:
I tried but can't get it to work, I have zero experience with Paraview. CSV files were only opened as table, and even after renaming to .xyz I don't get to see points.

What part of the point cloud am I looking at in your screenshots? My images were all top views.

vdsmax, 2020-04-17:
Once you open a csv file in Paraview, select the opened table in the pipeline browser and use the "Table to points" filter in the filters menu at the top of the Paraview window. Click on the result in the pipeline browser, set the XYZ colums, select the point representation and a color. By clicking on apply button, the point cloud will be display.
To move a PointClound in Paraview, you can use the filter "Transform". With this function you can compute the translation and rotation you want.

Also, the pmicp command should generates some vtk file of the registration in the path where the command is executed. You can also open these files in Paraview to see the different PointClouds: test_ref.vtk is the reference use for the registration (200402_075853_S1.csv in our case), test_data_in.vtk is the entry of the registration (for us 200402_102843_S1.csv) and test_data_out.vtk is the result of the registration.

The previous picture was about a corner of the dataset. Here is the results i have obtained for both poles with the view you did. As you can see, i don't have a big shift, especially for the pole 2 as you have.

[after_registration_base_pylon_1.pdf](https://github.com/ethz-asl/libpointmatcher/files/4490554/after_registration_base_pylon_1.pdf)
[after_registration_base_pylon_2.pdf](https://github.com/ethz-asl/libpointmatcher/files/4490553/after_registration_base_pylon_2.pdf)

tfwittwer, 2020-04-17:
Thanks a lot for your help, I really appreciate it. I got it to work now. I loaded the VTK files, and now the result does indeed match what you get. I'm now trying to understand why the translation vectors obtained by manual translation are different from what is reported in the complete_transfo.txt file.

[image]

[image]

I use only the translation part and discard the rotation matrix. Could this be the issue? I have my doubts considering the rotation angles are very small. Is the reported translation vector in a different coordinate system?

I'd like to eliminate the first option by trying the PureTranslation option. What would I have to add to the .yaml file to get this to work? I tried adding 

transformation:
  PureTranslation

to the top, but this doesn't seem to work.

tfwittwer, 2020-04-17:
I see that use of the transformations module is commented out in ICP.cpp. Any particular reason for that?

vdsmax, 2020-04-17:
Your welcome.
You should also use the rotation given by the registration. To use the result with Paraview, convert the rotation matrix obtained in XYZ Euler angle in degrees. Then add them in the rotation parameters. The result of the registration will be more accurate.
The transform given by icp is in the map reference frame, which are the XYZ axes of the reference point cloud in our case (200402_075853_S1.csv).

The PureTranslation option is not yet computed. That's why the transformations module is commented out in ICP.cpp. You can't use it witth the yaml file configuration.
If you want just the translation, you can compute the translation between the centroid of each Pointcloud. The Point-to-point minimization is doing this method.

tfwittwer, 2020-04-17:
That makes sense, thanks!

closed 2020-04-17 by tfwittwer