norlab-ulaval/libpointmatcher #380: Handling of timing information by DataPoints class

YoshuaNava, 2020-05-18:
Hi,
By peeking through the code I noticed that the [DataPoints class](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/PointMatcher.h#L207) has a [Time field](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/PointMatcher.h#L335).

The Time field is not documented in the [header section right above the class](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/PointMatcher.h#L195), and I couldn't find an example that shows how it is intended to be used.

By looking at the definition I infer that it is intended to hold the time per point in the point cloud, which is quite nice. It would allow people to undistort point clouds and even analyze which points were introduced when.

However, it is unclear:

* Why the type is Int64 instead of a standard C++ time type as std::chrono. (credits to [this GTSAM PR](https://github.com/borglab/gtsam/issues/315))
* If the time should be local (e.g. from start to end of point cloud capture) or global (e.g. Unix time).

By means of this issue I would like to start a discussion on this type and get more information so that we can improve the documentation / add an example or use a standard type.

pomerlef, 2020-05-19:
When looking at [std::chrono](https://en.cppreference.com/w/cpp/chrono), I see a lot of c++20. The implementation dates from a couple of years already. I essentially look at what ROS was providing, which is typically splitted Int64. See [ros::TimeBase](http://docs.ros.org/latest/api/rostime/html/classros_1_1TimeBase.html).

We can add a label to times, so one could have different representation (local or global). One could argue that we don't need int64 for local time. I could see the situation where one would like to track the name of the clock that produced those timestamps (i.e., in a global map with multiple lidar involved).

As for the hidden feature situation. I had to code something for a proof of concept a while ago and the branch was stale for too long. I merged it while I was still understanding what was going on hoping that someone pick it up. So here we are ;)

pomerlef, 2020-05-19:
I recall having a discussion about that with some early dev, but I don't find any trace of it. For now, there was a code review here #169 (just for general info).