norlab-ulaval/libpointmatcher #389: crossProduct in point_to_plane

EdwardLiuyc, 2020-06-19:
I've noticed the comment in the function `crossProduct`  

```//Note: A = [x, y, z, 1] and B = [x, y, z] for convenience```  

but I did not see any use of the homogenous coordinate?

pomerlef, 2020-06-19:
I'm not quite sure about the question, but `A` is expected to be with a dimension of 4 in 3D and 3 in 2D. It's only used to assert the proper dimension on line [290](https://github.com/ethz-asl/libpointmatcher/blob/b812b0958d3b65bad7c15716fe843fe752ad8d60/pointmatcher/ErrorMinimizer.cpp#L290), not for the computation.

EdwardLiuyc, 2020-06-20:
I noticed that too. Would 

```crossProduct(A.topRows(3), B.topRows(3))```

 be more reasonable ?  then the assert change to `assert(A.rows() == B.rows())`?

pomerlef, 2020-06-20:
It was a very specialized function coded to keep the math readable in `PointToPlaneErrorMinimizer<T>::compute_in_place`, with no real intention to be reused. As far as I can see, it is only used [here](https://github.com/ethz-asl/libpointmatcher/blob/42905f4d458a09ff691b3096c4b206c2170cac58/pointmatcher/ErrorMinimizers/PointToPlane.cpp#L200).

The call would need to change to `cross = this->crossProduct(mPts.reading.features.topRows(3), normalRef);`. I don't recall if 'topRows()` copy data or reference data.

That being said, I'm not sure what is the motivation for this refactoring. Do you need `crossProduct` somewhere else?

EdwardLiuyc, 2020-06-21:
I don't need use it somewhere else and the only thing I want to say is that this function is confusing. And I got it that it is just for efficency. 
Close it. Thanks anyway.

closed 2020-06-21 by EdwardLiuyc