norlab-ulaval/libpointmatcher #390: Cannot grep version number: The system cannot find the file specified

npd26, 2020-07-17:
Hello,

I am getting this error while following step from tutorial to build libnabo. Any idea how to solve this problem?
windows 10, cmake 3.16.2

cmake error

```

CMake Error at CMakeLists.txt:20 (message):
  Cannot grep version number: The system cannot find the file specified

Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
OpenCL disabled, not looking for it
ANN not found, disabling it in benchmarks
FLANN not found, disabling it in benchmarks
Could NOT find PythonInterp (missing: PYTHON_EXECUTABLE) 
Python version from interpreter is 
Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) 
Python libs or executable not found, skipping Python bindings
Configuring incomplete, errors occurred!
See also "C:/VC/libnado_build/CMakeFiles/CMakeOutput.log".
```

also CMakeOutput.log showing not any error as mentioned in [https://github.com/ethz-asl/libpointmatcher/issues/118](url)

thanks.

pomerlef, 2020-07-17:
You might have missed this steps:

>Prerequisites
If your operating system does not provide it, you must get Eigen, and Boost if you want to build the Python bindings. Eigen only needs to be downloaded and extracted. **You also need grep**, which is available in standard on Linux or Mac OS X, you can get the window version here http://gnuwin32.sourceforge.net/packages/grep.htm.

npd26, 2020-08-06:
after adding grep for the windows and set it up in system variables, still I am getting same error.

closed 2020-08-06 by npd26

danielaDESM, 2021-03-19:
Probably to  late the answer, but in my case it was solved by following the following post:
https://github.com/ethz-asl/libpointmatcher/issues/118#issuecomment-676490427
It is apparently something related to the CMakeLists of libnabo.