norlab-ulaval/libpointmatcher #393: Hotspots in Random Sampling Filter

YoshuaNava, 2020-07-23:
As part of my efforts to benchmark Libpointmatcher, I created a toy example that processes point clouds coming from a ROS topic. 

I ran the node through Intel VTune, for 100 seconds, with 96 of them being actually dedicated to processing, during the rest the node was idle.

In the following lines I describe my search for hotspots and a short analysis.

## Hotspots

### CPU

[image]

### Memory access

[image]

### Processor events

[image]

*Note*: there are lots of misaligned memory moves shown in the above figures. A big part of those is due to ROS deserialization (TCPRead), another part to libpointmatcher_ros deserialization (reported [here](https://github.com/norlab-ulaval/libpointmatcher_ros), and another part to the way libpointmatcher not using move semantics -I think-). They are not the core of this issue.

## Analysis

The call to std::random in the random sampling data points filter happens within the main sampling loop:

[image]

Sampling a random number is a fast operation, but the way it is currently implemented defaults to a scalar operation, probably preventing the loop from getting vectorized. Furthermore, it is retiring (running through the CPU execution units) a lot of times, when compared with other instructions. The average latency of a single random call is 23 cycles, almost as much as the calls to the kernel to switch threads / reserve memory.

YoshuaNava, 2020-07-23:
## Proposals

I would propose to implement a similar solution as the one presented in #328, where we would:

* Bring a generator for a uniform distribution (please confirm if this is OK).
* Create a vector of indices within the range of a point cloud.
* Random shuffle the vector indices.
* Keep only X percent of the columns after random sampling.

@pomerlef Do you think this would be a good solution? Is there any specific test case you would like me to look at?

pomerlef, 2020-07-23:
Good catch! I'm not quite sure about your test set up, but it could be useful to run the analysis on the unit test as they apply registration using bunch of filters without passing through ROS.

I was under the impression that `std::rand()` was uniform. If that was not the case and you're converting it to uniform, that's even better.

The proposal looks fine for me. The `RandomSamplingDataPointFilter` is used quite a lot in general. Just make sure to visually assess the filtered point cloud before and after to make sure that there no artefact in the point cloud after the modification.

YoshuaNava, 2020-07-25:
Hi @pomerlef 
Thank you for your prompt reply.

I tested the solution proposed (see code snippet below) and found it to be way slower than expected. Due to multiple reasons:

1) Creating a vector of ints is expensive in itself, O(n) worst case.
2) Shuffling the vector sequentially is O(n) as well, even with a random number generator.
3) Shuffling doesn't seem to run in parallel? (the compiler -gcc does not seem to auto-vectorize the code)

```cpp
  Eigen::VectorXi indices = Eigen::VectorXi::LinSpaced(nbPointsIn, 0, nbPointsIn - 1);
  std::random_device randomDevice;
  std::minstd_rand randomGenerator(randomDevice());
  std::shuffle(indices.data(), indices.data() + nbPointsIn, randomGenerator);
  Eigen::Index j = 0;
  for (Eigen::Index i = 0; i < nbPointsOut; i++) {
    // cloud.setColFrom(indices[j], cloud, i);
    cloud.swapCols(indices[i], j);
    j++;
  }
  cloud.conservativeResize(j);

```

YoshuaNava, 2020-07-25:
Now, I implemented two other solutions that did work, and speeded up the filter:

## Solution 1

Sampling floats in the range [0,1] from a uniform distribution using the std::random objects. Removed random sampling from the actual column-swapping loop.

Increases the speed in around 25-30%

```
	std::random_device rd;
  	std::minstd_rand gen(rd());
  	std::uniform_real_distribution<float> dis(0, 1);
	const Eigen::VectorXf randomNumbers = Eigen::VectorXf::NullaryExpr(nbPointsIn, [&](float dummy){return dis(gen);});
	int j=0;
	for (Eigen::Index i = 0; i < nbPointsIn && j<=nbPointsOut; i++) {
		if (randomNumbers[i] < prob)
		{
			cloud.setColFrom(j, cloud, i);
			++j;
		}
	}
	cloud.conservativeResize(j);

```

## Solution 2

Using a random generator from std::random, but without sampling from a uniform distribution. All random samples are saved in an Eigen vector before our column-swapping sampling loop.

```
	std::random_device rd;
  	std::minstd_rand gen(rd());
	const Eigen::VectorXf randomNumbers = Eigen::VectorXf::NullaryExpr(nbPointsIn, [&](float dummy){return gen();});
	int j=0;
	for (Eigen::Index i = 0; i < nbPointsIn && j<=nbPointsOut; i++) {
		if (randomNumbers[i] < prob)
		{
			cloud.setColFrom(j, cloud, i);
			++j;
		}
	}
	cloud.conservativeResize(j);
```

pomerlef, 2020-08-02:
I could worth it to try the [Marsaglia's xorshf generator](https://stackoverflow.com/questions/1640258/need-a-fast-random-generator-for-c).

For solution 2, if it's not uniform, do we know what the histogram looks like? There might be some implication on the resulting density of points. I doubt, I would code both and use a parameter `approximation=true` to activate solution 2.

YoshuaNava, 2021-02-26:
Comparison of proposed methods in the [pull request](https://github.com/ethz-asl/libpointmatcher/pull/396#issuecomment-786647838)

@pomerlef I just read your request about testing Marsaglia's generator. It is even faster than the proposed <random> methods

[image]

This method is approx. 1.5x faster than the ones I proposed, but the distribution of values looks slightly less random. Note that I'm sampling float values from xorshf96 by dividing over the max value of unsigned long: https://github.com/raylee/xorshf96/blob/master/xorshf96.c#L23

See attached files for the float values sampled by the original, proposed (direct RNG and uniform), as well as xorshf96.

[original.txt](https://github.com/ethz-asl/libpointmatcher/files/6050293/original.txt)
[direct_rng.txt](https://github.com/ethz-asl/libpointmatcher/files/6050291/direct_rng.txt)
[uniform.txt](https://github.com/ethz-asl/libpointmatcher/files/6050292/uniform.txt)
[xorshf96.txt](https://github.com/ethz-asl/libpointmatcher/files/6050294/xorshf96.txt)

closed 2021-02-26 by pomerlef