norlab-ulaval/libpointmatcher #402: Sphericality.h errors when building on windows

dagata-mining, 2020-08-26:
When building on windows, I had errors with sphericality.h (may be related to recent commit commit 9f4127b112f9af4074b7647df561072c54d2abfd ). I had to get rid of all dependencies with sphericality and the build did work on windows.

pomerlef, 2020-09-14:
Could you post the complet error?

pomerlef, 2020-09-14:
@kubelvla please have a look once the details of error is available.

dagata-mining, 2020-09-14:
So first I had dependency errors (looks like a problem when parsing C:/Program Files/libnabo.. with the space between Program&Files when cmake is building the solution probably) So I modified the Linker->Input->Additional dependencies of each modules manually on visual studio.:

```
Error LNK1104 cannot open file 'C:\Program.obj' pmicp C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' list_modules C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' icp_advance_api C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' compute_overlap C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' icp_customized C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' icp_simple C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' build_map C:\Program Files\libpointmatcher\build\examples\LINK 1
Error LNK1104 cannot open file 'C:\Program.obj' align_sequence C:\Program Files\libpointmatcher\build\examples\LINK 1

```
So I modified the path that was: 
**C:\Program
Files/libnabo/build/RelWithDebInfo/nabo.lib.lib**

to 

**C:\Program Files\libnabo\build\RelWithDebInfo\nabo.lib**

And that's when I got the sphericality.h errors (my quick fix was to remove any dependecies of sphericality.h to all the files). Here's the error (got them for each modules.. compute_overlapp icp_custom etc..)

```
Error LNK1120 2 unresolved externals compute_overlap C:\Program Files\libpointmatcher\build\examples\RelWithDebInfo\compute_overlap.exe 1
Error LNK2019 unresolved external symbol "public: __cdecl SphericalityDataPointsFilter<float>::SphericalityDataPointsFilter<float>(class std::map<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct std::less<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > >,class std::allocator<struct std::pair<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const ,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > const &)" (??0?$SphericalityDataPointsFilter@M@@QEAA@AEBV?$map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@U?$less@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@@std@@@2@@std@@@Z) referenced in function "public: virtual class std::shared_ptr<struct PointMatcher<float>::DataPointsFilter> __cdecl PointMatcherSupport::Registrar<struct PointMatcher<float>::DataPointsFilter>::GenericClassDescriptor<struct SphericalityDataPointsFilter<float> >::createInstance(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::map<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct std::less<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > >,class std::allocator<struct std::pair<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const ,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > const &)const " (?createInstance@?$GenericClassDescriptor@U?$SphericalityDataPointsFilter@M@@@?$Registrar@UDataPointsFilter@?$PointMatcher@M@@@PointMatcherSupport@@UEBA?AV?$shared_ptr@UDataPointsFilter@?$PointMatcher@M@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@5@AEBV?$map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@U?$less@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@@std@@@2@@5@@Z) compute_overlap C:\Program Files\libpointmatcher\build\examples\pointmatcher.lib(Registry.obj) 1
Error LNK2019 unresolved external symbol "public: __cdecl SphericalityDataPointsFilter<double>::SphericalityDataPointsFilter<double>(class std::map<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct std::less<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > >,class std::allocator<struct std::pair<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const ,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > const &)" (??0?$SphericalityDataPointsFilter@N@@QEAA@AEBV?$map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@U?$less@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@@std@@@2@@std@@@Z) referenced in function "public: virtual class std::shared_ptr<struct PointMatcher<double>::DataPointsFilter> __cdecl PointMatcherSupport::Registrar<struct PointMatcher<double>::DataPointsFilter>::GenericClassDescriptor<struct SphericalityDataPointsFilter<double> >::createInstance(class std: [... cut ...] intMatcherSupport@@UEBA?AV?$shared_ptr@UDataPointsFilter@?$PointMatcher@N@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@5@AEBV?$map@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@U?$less@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@V?$allocator@U?$pair@$$CBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V12@@std@@@2@@5@@Z) compute_overlap C:\Program Files\libpointmatcher\build\examples\pointmatcher.lib(Registry.obj) 1

```

kubelvla, 2020-09-15:
I'll try to compile the library on windows to see.

kubelvla, 2020-10-03:
Ok, that was painful, mainly CMaking new Libnabo. Respect @dagata-mining :D
Anyway, can you please help me verify my fix:
\libpointmatcher\pointmatcher\DataPointsFilters\Sphericality.cpp line 142,143: 
   in if condition, replace the two "or" by "||"

pomerlef, 2020-11-25:
It seems that the keyword `and` is also present somewhere. Someone send me the following compiling errors:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C3927	'->': trailing return type is not allowed after a non-function declarator	pointmatcher	D:\GIT\libpointmatcher\pointmatcher\DataPointsFilters\SpectralDecomposition.cpp	278	
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2065	'and': undeclared identifier (compiling source file D:\GIT\libpointmatcher\pointmatcher\DataPointsFilters\Saliency.cpp)	pointmatcher	D:\GIT\libpointmatcher\pointmatcher\DataPointsFilters\utils\sparsetv.hpp	403	
```