norlab-ulaval/libpointmatcher #403: Alignment confidence indication

Gerharddc, 2020-09-07:
I am trying to align 2D LIDAR scan frames to frames in a database to get a position estimate. I am using image matching to suggest candidates for this, but there is a chance that some suggestions might be complete garbage or that the environment will have changed dramatically from the LIDAR's perspective to the point where alignment will not work anymore, so I want to find a way to filter out point cloud alignments that make no sense at all. Unfortunately it seems that the icp compute function does not return any sort of metrics that I can attempt to use for this purpose. As far as I can determine the function only returns the final transformation matrix estimate, but I would like to find out if there is some way that I can access to for instance the final error value or the amount of inliers or any other useful metrics of the alignment quality?

pomerlef, 2020-09-09:
What you are looking for is a covariance for the outputted transformation. Unfortunately, there is no functional way to compute it (you can search for covariance in the issues of this library to see the different discussion on that topic.

To get closer to what you want, you can have a look at residual errors after the minimization. From the documentation:
> How to compute the residual error with ErrorMinimizer::getResidualError(...) [See the example code provided here.](https://github.com/ethz-asl/libpointmatcher/issues/193#issue-203885636)

Gerharddc, 2020-09-10:
@pomerlef thanks I think the residual errors are indeed what I wanted. The example code is a bit strange though in that it seems that one has to force some calculations that I can only assume would already have been done as part of the optimisation process already so I assume this is not the most efficient approach possible but it does seem to work.

YoshuaNava, 2020-09-14:
@pomerlef Something I have observed in the past is that the residual error value seems to be tied to the relative number of points in the reference and sensor point cloud. Is there a sound way to normalize this quantity?

pomerlef, 2020-09-14:
By looking at what is implemented [over here](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/ErrorMinimizers/PointToPlane.cpp#L350), it seems that the function returns the sum of the residuals. I don't recall the motivation behind that choice.

You would need to divide by the number of points to get the average residual error.