norlab-ulaval/libpointmatcher #405: Build issues "Linking CXX executable icp_customized"

ArvidPeterson, 2020-09-23:
I am running into problems when I run make in the /build directory following [this](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationUbuntu.md) tutorial (end of step 3). I am building libpointmatcher for use in rtabmap_ros.

The following is printed from the utest/listVersionsUbuntu.sh:
```

-e Name 		| Version
-e ----------------|-------------------------------
-e ubuntu: 	| Description: Ubuntu 18.04.5 LTS
-e architecture: 	| 64-bit
-e gcc: 		| gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
-e git: 		| git version 2.17.1
-e cmake: 		| cmake version 3.10.2 CMake suite maintained and supported by Kitware (kitware.com/cmake).
-e boost: 		| Version: 1.65.1.0ubuntu1
-e eigen3: 	| Version: 3.3.4-4
-e doxygen: 	| Version: 1.8.13-10
-e 

```
The following error is printed when running make:

```
[ 77%] Built target yaml-cpp-pm
[ 77%] Built target pointmatcher
[ 77%] Built target gtest
[ 77%] Linking CXX executable icp_customized
../libpointmatcher.a(MatchersImpl.cpp.o): In function `MatchersImpl<float>::KDTreeMatcher::init(PointMatcher<float>::DataPoints const&)':
/home/arvid/Libraries/libpointmatcher/pointmatcher/MatchersImpl.cpp:82: undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
../libpointmatcher.a(MatchersImpl.cpp.o): In function `MatchersImpl<double>::KDTreeMatcher::init(PointMatcher<double>::DataPoints const&)':
/home/arvid/Libraries/libpointmatcher/pointmatcher/MatchersImpl.cpp:82: undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
../libpointmatcher.a(MatchersImpl.cpp.o): In function `MatchersImpl<double>::KDTreeVarDistMatcher::init(PointMatcher<double>::DataPoints const&)':
/home/arvid/Libraries/libpointmatcher/pointmatcher/MatchersImpl.cpp:129: undefined reference to `Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<double, Eigen::Matrix<double, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
../libpointmatcher.a(MatchersImpl.cpp.o): In function `MatchersImpl<float>::KDTreeVarDistMatcher::init(PointMatcher<float>::DataPoints const&)':
/home/arvid/Libraries/libpointmatcher/pointmatcher/MatchersImpl.cpp:129: undefined reference to `Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::create(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, Nabo::NearestNeighbourSearch<float, Eigen::Matrix<float, -1, -1, 0, -1, -1> >::SearchType, unsigned int, Nabo::Parameters const&)'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/icp_customized.dir/build.make:104: recipe for target 'examples/icp_customized' failed
make[2]: *** [examples/icp_customized] Error 1
CMakeFiles/Makefile2:346: recipe for target 'examples/CMakeFiles/icp_customized.dir/all' failed
make[1]: *** [examples/CMakeFiles/icp_customized.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

any help is very much appreciated!

//Arvid P

pomerlef, 2020-09-23:
This seems to be related to the ongoing PR #382. We are working on it.

ArvidPeterson, 2020-09-23:
Cheers! I appreciate the quick answer.

ArvidPeterson, 2020-09-24:
Solved by pr [#382](https://github.com/ethz-asl/libpointmatcher/pull/382) as suggested by @pomerlef. Closing it. Cheers!

closed 2020-09-24 by ArvidPeterson