norlab-ulaval/libpointmatcher #409: Transformation on point cloud throws an error

dhuettig, 2020-10-19:
Hi,
I'm using the provided [example code](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Transformations.md) to test a transformation on a point cloud but I get the runtime error `Bus error: 10`. I'm using the same file `examples/data/car_cloud400.csv` as in the example description. 

I had to change the line `rigidTrans = PM::get().REG(Transformation).create("RigidTransformation");` to `rigidTrans = PM::get().REG(Transformation).create("RigidTransformation").get();`, otherwise the compiler throws the following error: 
```cpp
error: assigning to 'PM::Transformation *' from incompatible type 'std::shared_ptr<Transformation>'
        rigidTrans = PM::get().REG(Transformation).create("RigidTransformation");
```

pomerlef, 2020-10-19:
@AlexandreG87 or @simonpierredeschenes is that caused by the latest change for shared pointers?

@dhuettig The correction is only to change to code example? If that is the case 5a72892 should resolve the issue.

AlexandreG87, 2020-10-19:
@pomerlef  It seems that the example code in the documentation hasn't be updated when @simonpierredeschenes change the raw pointers to shared pointers in this PR #284. So, for now, I suggest @dhuettig to change `PM::Transformation*` to `std::shared_ptr<PM::Transformation>` instead.

I'll fix the doc and submit a PR.

dhuettig, 2020-10-19:
Thank you for the replies.

@pomerlef using ```PM::get().REG(Transformation).create("RigidTransformation").get();``` doesn't resolve the issue since then the compiler throws a ```bus error 10```.

@AlexandreG87 using ```std::shared_ptr<PM::Transformation>``` worked and the transformation can be applied.

closed 2020-10-19 by dhuettig