norlab-ulaval/libpointmatcher #420: Windows Compiling error due to using  "and" and "or" and "not" as logical operators instead of using the &&, || and ! notation

princetn, 2020-11-25:
There are a couple of files which use  "and" and "or" and "not" as logical operators instead of using the &&, || and ! notation. This is an archaic syntax which will prevent someone using Windows from compiling libpointmatcher.  
here is what the errors would look like:
Here is the first error:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C3927	'->': trailing return type is not allowed after a non-function declarator	pointmatcher	D:\GIT\libpointmatcher\pointmatcher\DataPointsFilters\SpectralDecomposition.cpp	278	
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2065	'and': undeclared identifier (compiling source file D:\GIT\libpointmatcher\pointmatcher\DataPointsFilters\Saliency.cpp)	pointmatcher	D:\GIT\libpointmatcher\pointmatcher\DataPointsFilters\utils\sparsetv.hpp	403

yokoshou, 2020-11-30:
/*quick manual*/
0.Open Project Property.
1.Click the C/C++ folder.
2.Click the Advanced property page.
3.Modify the Forced Include File property. (add iso646.h)

/*Detail about this problem*/

I was in the same problem and I research about it.

It may be caused by Visual Studio Language Extensions.

If you just want to disable this function, you need to 
1.choose Configuration Properties > C/C++ > Language.
2.modify the Disable Language Extensions property.

But it doesn't work in this case.
Because the Boost thread function needs Language Extensions.

So another way to solve this problem is to add #include<iso646.h>.
I'm tired of searching which file needs to <iso646.h>, so I using Name Forced Include File.

1.Click the C/C++ folder.
2.Click the Advanced property page.
3.Modify the Forced Include File property. (add iso646.h)

good luck!

maximecharriere, 2020-12-09:
Thank's @yokoshou, it worked for me ! 
Can't we make this mistake not happen anymore?

closed 2020-12-14 by pomerlef