norlab-ulaval/libpointmatcher #425: Running utest failure

ryleymcc, 2020-12-04:
following [this](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationUbuntu.md#possible-caveats) tutorial to install libpointmatcher. 

when i run 

ryley@ryley-GL65-9SD:~/Libraries/libpointmatcher$ utest/utest --path ${SRC_DIR}/examples/data/

i get this

terminate called after throwing an instance of 'std::runtime_error'
  what():  Cannot open file /examples/data/2D_oneBox.csv
Aborted (core dumped)

>

ryleymcc, 2020-12-04:
i should add that this is a new install of Ubuntu 20.04 LTS

pomerlef, 2020-12-09:
What is in your variable `${SRC_DIR}`?

If you want you can navigate in the repo and search for the folder `data` and use that path after the flag `--path`. The folder exist here: https://github.com/ethz-asl/libpointmatcher/tree/master/examples/data

AlexandreG87, 2021-01-13:
Hi @ryleymcc! Did you manage to run the tests? If not, you must be in the build directory in order to be able to execute that command. You can also replace the `${SRC_DIR}/examples/data/` part by `../examples/data/`, since the variable `SRC_DIR` needs to be define every time a new terminal is opened.

ryleymcc, 2021-01-13:
I have switched my projects strategy and this is no longer an issue of mine. I will close this issue now.

closed 2021-01-13 by ryleymcc